Snapmaker/Luban #1348: Bug: Job Status not updating after connecting to running 3D print.

imaplt, 2021-12-13:
# 🐞 bug report

### Affected Version(s)
Version 4.1.1

### Is this a regression? (optional)
Unknown

### To Reproduce
Steps to reproduce the behavior:
1. Go to workspace.
2. Click on Connect while 3D print is in progress.
3. Observer Job Status pane on right.

When connecting to a running 3D project the connection will indicated a green dot next to 3D printing and Heated Bed. The job status however never changes. It DOES show the Stop and Pause buttons. It does NOT show any of the stats. Percent complete stays at 0% etc.

Expected behavior
I would expect that since the other indicators are there and appear to work that the status would as well. 

**Platform:**
- Luban 4.1.1
- macOS Monterrey
- Printer: Snapmaker 2 A350

sumibin, 2021-12-14:
The features of the working state need to be used with the new firmware. Have you upgraded the firmware of the machine to v1.14.1?

imaplt, 2021-12-14:
It is at 1.13. I did a check for updates and it says I have the latest version. I will upgrade now and the follow up.

imaplt, 2021-12-14:
I did the upgrade and it shows in the about info. The firmware option still shows the old one FYI. Testing status next.

imaplt, 2021-12-14:
Ok. I looks like the status is updating now. The file name says "Unknow". At a minimum this is a spelling error and should show the file name IMO since it is NOT unknown.

Also confirmed that the software update button in the firmware still shows as the 1.13 is the latest and installed per the screen and pressing check for updates button says the latest is installed.
 The about DOES show the correct firmware version.

Galindorf, 2021-12-21:
So I tried to update the firmware on my Snapmaker A350 after updating Luban, (as the updated Luban tells you to so). When I did it told me that it had the latest version installed, but my version was 1.13.

However! Yesterday I did a firmware upgrade check on a whim and it downloaded the update. Perhaps try it again. I am not sure why it was not giving me the update at first.

sumibin, 2022-03-24:
Luban v4.1.4 improvement: Screen sync job status with Luban

closed 2022-03-24 by sumibin