Snapmaker/Luban #1349: Bug: Workspace view/preview shows only inverted cone after update to 4.1.1

imaplt, 2021-12-13:
# 🐞 bug report

### Affected Version(s)
4.1.1

### Is this a regression? (optional)
Worked in the previous release version

### To Reproduce
Steps to reproduce the behavior:
1. Import object
2. Conver to GCode
3. Sent to workspace
2. Look at the view.

When I imported a new object, converted it and sent it to the workspace it shows up as an inverted cone. I then opened a previous project that was fine and it did the same thing,

Expected behavior
I would see the 3D view of the object instead of the cone.

## 🌍  Your Environment

**Platform:**
macOS Monterey
- Printer: Snapmaker A350
[image]

sumibin, 2021-12-14:
The preview function of G code is migrated to the preview button. The canvas in the center of the workspace is only used as a preview of the G code in printing.
[image]

closed 2021-12-14 by sumibin

imaplt, 2021-12-14:
Just to confirm that in this release, when exporting g-code to workspace I have to then click the preview to see it? Why the change? What does the code represent then? If I start printing then the workspace will show the print as it happens?

I guess the real issue is why the change.

Thanks!