Snapmaker/Luban #1365: Bug: Lubin didn't know a job was going on when I connected to it (in 3d printing)

kalmdown, 2021-12-20:
# 🐞 bug report

### Affected Version(s)
4.1.0

### Is this a regression? (optional)
Not sure. Didn't try and don't have an old version on hand.

### To Reproduce
Steps to reproduce the behavior:
I connected to SM via Luban. Machne was warming up for a print I had started. When I went to machine the dialog for accepting a connection was on screen and so I hit Yes. 

Exception or Error (optional)
Neither Luban or machine seem to know I am connected. Luban acts like the machine is available and doing nothing.

Expected behavior
I'd expect the machine to indicate it is connected and Luban to show me what its state is - in this case some %age along, doing a print. And expect to show progress as it goes. Then I'll know when it is done and be able go see what wonderful things have been made.

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: Windows 11
- Printer: Snapmaker A350

</code></pre>

sumibin, 2022-01-06:
As this is a new feature, firmware upgrade 1.14.1 is required.

sumibin, 2022-01-06:
You can try the firmware version 1.14.1. This version of the firmware works with 4.1 Luban to know the status of the job through the job status Widget.

closed 2023-03-21 by sumibin