Snapmaker/Luban #1372: Bug: Cannot set material thickness in laser

workingmatt, 2021-12-23:
When laser cutting I cannot find anywhere in Luban or via the touchscreen to set the thickness of the material. This was done in Luban's previous versions.

With the latest version I cannot find any way to set the thickness of the material. All I can find is via the touchscreen Settings - Laser - Adjust Laser Height. Surely this is a mistake!?

# 🐞 bug report

### Affected Version(s)
Luban version 4.1.1
Snapmaker 2.0 A350 Firmware Snapmaker2_V1.14.1_20211207

### Is this a regression? (optional)
Yes, this is a regression. Earlier versions of Luban allowed you to set the thickness of the material although this workflow was clunky and I often forgot to set it correctly.

### To Reproduce
Steps to reproduce the behavior:
Connect to a laser configured Snapmaker 2
Try to set the material thickness - used to be in the workspace.

Expected behavior
A material thickness dialogue should be available somewhere.

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: Windows 10 x64
- Printer: Snapmaker 2.0 A350
- Firmware Snapmaker2_V1.14.1_20211207
</code></pre>

jane-rose, 2021-12-24:
Once you click the `start print` button, you can see the pop-up window for setting material thickness.

workingmatt, 2021-12-28:
I've found that. Setting the material thickness and clicking Start does work and starts the job. However this is also the first time we get to see a 'preview' of the cut on the workspace grid. There is no opportunity to check the cut before starting it.
The pop-up asking for material thickness is an improvement to the workflow because one cannot forget to set it. The trigger to open that dialog should be a 'Show preview' or similar rather than a Start Job to allow the user to review.

closed 2022-01-06 by jane-rose