Snapmaker/Luban #1374: Bug: xxxx in 3d printing

climatec, 2021-12-27:
# 🐞 bug report

### Affected Version(s)

 The issue is caused by version x.y.z

### Is this a regression? (optional)

 Yes, the previous version in which this bug was not present was: ....

### To Reproduce
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error
 A clear and concise description of the problem...

Exception or Error (optional)
<pre><code>

</code></pre>

Expected behavior

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: [Windows 10 x64, macOS Big Sur, Ubuntu 20.04, etc]
- Printer: [Snapmaker Original, Snapmaker A150, etc]

</code></pre>

closed 2022-01-06 by jane-rose

climatec, 2022-01-06:
Snapmaker, why is this closed without any comment????

jane-rose, 2022-01-06:
> Snapmaker, why is this closed without any comment????

Sorry, on our end, we don’t see any descriptions, please check if any descriptions are written.

climatec, 2022-01-06:
Sorry i don’t know why this 1374 are made, the right one is 1375.  I apologise. Just delete this one 1374.