Snapmaker/Luban #1375: Bug: Z hop height in 3d printing

climatec, 2021-12-27:
#
[image]

Hi Snapmaker team. 

3D printing problem 
When trying to activate 1mm Z hop height in luban I get the error that Generated G-code stepped out of the cube.  
the model I am printing is just 3 mm high. So there must be a bug 🐞 in that part of the program. 

# 🐞 bug report

### Affected Version(s)

 The issue is caused by version x.y.z

### Is this a regression? (optional)

 Yes, the previous version in which this bug was not present was: ....

### To Reproduce
Steps to reproduce the behavior:
1. Go to 'luban choose 3D printing open a small item to print'
2. Click on 'Z hop height and set to 1mm.'
3. Click on  'click on generate G code '
4. See error
 A clear and concise description of the problem...

Exception or Error (optional)
<pre><code>

</code></pre>

Expected behavior
Luban should be accepting a Z hop on 1mm on a item only 3mm height

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: macOS Big Sur
- Printer: Snapmaker A350

</code></pre>

harry-png, 2021-12-29:
Same problem with moving object to a different location 
luban 4.1.3
mac os Monterey version 12.0.1

apper002, 2022-01-02:
same here 4.1.3

again another bug.........

jane-rose, 2022-01-11:
@climatec  On the way, it will be repaired on v4.1.4. (Expected on Jan 14)

jane-rose, 2022-01-14:
#1396

jane-rose, 2022-01-17:
Please check this release https://github.com/Snapmaker/Luban/releases/tag/v4.1.4. Any feedback can be raised in this issue。

closed 2023-03-21 by sumibin