Snapmaker/Luban #1376: Feature Request: 4-axis Job Setup

jwrodrick, 2021-12-27:
Currently, when configuring a 4-axis job we provide the length and diameter of the material. This does not allow for cylindrical objects that change diameters throughout the length of the material. Think of a Yeti Tumbler for example, at the top it's 100mm and at the bottom it's 70mm. I have no way to change the Z-axis height along the length of the material to ensure the laser (in this case) remains the same height down the length. This results in the laser becoming unfocused as it moves down the length of the cup and since the last firmware update we've lost the ability to manually change the Z-axis start point.

I propose that you add the ability for the user to add multiple diameter settings at points along the length of the material. The software should calculate the slope of the Z-axis along the points of reference and make the changes to the g-code. 

Examples of the data provided by user:
Length: 200mm
Diameter at top: 100mm (single input)
Diameter at bottom: 70mm (single input)
Point of reference diameter: distance from top: 100mm & diameter at point: 85mm (2 inputs, this should be a data point that we can add as many times as we need to show multiple changes to the diameter across the length)

sumibin, 2022-01-06:
About "since the last firmware update we've lost the ability to manually change the Z-axis start point.
":The firmware does not remove the ability to manually set the origin, can you describe it in more detail

jwrodrick, 2022-01-06:
On the touch screen, the button to jog on the z-axis is disabled. Also,
swiping right the button to set z-axis is disabled. I can provide
screenshots tomorrow.

closed 2023-03-21 by sumibin