Snapmaker/Luban #1384: Unable to set nozzle or bed temperature via touch screen

DirkM20, 2021-12-30:
Affected Version(s) firmware 1.14.1

The issue is caused by version 1.14.1
Is this a regression? works fine in older firmware

Yes, the previous version in which this bug was not present was: 1.13.2

Steps to reproduce the behavior:
start print job
change nozzle or bed temp

Exception or Error
nozzle or bed temp does not change

Expected behavior
nozzle or bed temp changes to desired value

setting nozzle or bed temp via octoprint works fine, suppose issue with touch panel to controller communication

brprigge, 2022-01-05:
Having a similar issue with 1.14.1. None of the linear modules move. Was not an issue on 1.12

parachvte, 2022-12-22:
@brprigge @DirkM20 Is this issue fixed in latest version?

closed 2022-12-23 by parachvte