Snapmaker/Luban #1394: Bug: Rectangles with rounded corners in SVG result in wrong vector toolpath for laser in Luban 4.1.3

caerandir, 2022-01-04:
# 🐞 bug report

Luban seems to ignore rounded rectangle corners in a SVG when calculating the vector toolpath. The display of the imported SVG itself looks right, only the toolpath is wrong. Workaround that works: Convert the rectangle with rounded corners into a path (e.g. in Inkscape). After that, all is correct.

### Affected Version(s)
Luban 4.1.3

### To Reproduce
Steps to reproduce the behavior:
1. Go to Laser - 3 axis
2. Open attached SVG file - Result: OK, rectangles have rounded corners
[image]
3. Create Vector toolpath: 
[image]
4. Preview Toolpath - Result wrong - rectangles have no rounded corners
[image]

Expected behavior
The toolpath should honor the rounded corners of the two rectangles.

## 🌍  Your Environment

**Platform:**
- Operating System: Windows 10 x64
- Printer: Snapmaker A350
- Luban 4.1.3

Attached ZIP contains:
* RectangleRoundedCorners.svg --> Inkscape created SVG with two rectangles with rounded corners and a bit more
* ConvertedToPath.svg --> Rectangles converted to Pathes. If you load this and run above steps, the result is correct:
[image]

[Testfiles.zip](https://github.com/Snapmaker/Luban/files/7809946/Testfiles.zip)

theevl, 2022-02-24:
I just encountered this problem.
it appears that the preview of the TOOLPATH predicts the "square corners" but the SIMULATION shows rounded corners.
I am using Luban 4.14 on Windows 10

jane-rose, 2022-02-24:
OK, since we reproduce this on v4.1.4. But we had sloved it on [v4.2.1-beta.1](https://github.com/Snapmaker/Luban/releases/tag/v4.2.1-beta.1). You can use Luban v4.2.1-beta.1 or wait for the official version of the subsequent v4.2.

closed 2023-03-21 by sumibin