Snapmaker/Luban #1400: Bug: in Ver 4.1.3 - Nozzle & Bed temperature varies from set temperature on Luban

sibiinhere, 2022-01-08:
# 🐞 bug report

### Affected Version(s)

 The issue is caused by version 4.1.3

### Is this a regression? (optional)

 Yes, the previous version in which this bug was not present was: 3.15.2

### To Reproduce
Steps to reproduce the behavior:
1. Go to 'Luban Software V4.1.3'
2. Click on 'Materials'
3. Click on 'Material Settings'
4. Create a Copy of PLA and rename it as 'PA6+30GF'
5. Set Printing Temperature as '270'
6. Set Initial Layer Printing Temperature as '275'
7. Set Final Printing Temperature as '270'
8. Set Heat Bed Temperature as '65'
9. Set Heat Bed Temperature as '75'
10. Scroll down & click on 'Select'
11. Add & upload a desired Design to 3D Print
12. Generate G code
13. Load the PA6+30GF Material
14. Connect to wifi & load the G-Code to the 3D Printer
15. When the printer starts to print
16. The displayed set Temperature is only '215'
17. Error - Material doesn't reach the desired temperature & hence there is no material flow.
 The material prints only at 275 dec C. But, the printer starts printing at 215 deg C itself.... Hence, there is no Material flow from the Nozzle.

Exception or Error (optional)
<pre><code>

</code></pre>

Expected behavior

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: [Windows 8.1 x64]
- Printer: [Snapmaker A350]

</code></pre>

deimi, 2022-01-16:
A workaround could be:
- Select a different material then the default one after start
- Generate code
- Close preview
- Select your desired material
- Generate code

jane-rose, 2022-01-17:
> A workaround could be:
> 
> 1. Select a different material then the default one after starting Luban
> 2. Generate code
> 3. Close preview
> 4. Select your desired material
> 5. Generate code

Sorry, we can't reproduce it. Please send your `def.json` file or confirm whether the exported file is already incorrect

closed 2023-03-21 by sumibin