Snapmaker/Luban #1406: Bug: PNG preview very inaccurate for laser

F1nnM, 2022-01-11:
# 🐞 bug report

### Affected Version(s)
I'm using 4.1.3, I assume most versions.

### To Reproduce
Steps to reproduce the behavior:
1. Create a svg
2. Export that svg to a png
3. Load both into luban, (evtl. set sizes to match)

Problem:
The previes don't match in scale. The png preview appears to be larger. By generating some gcode, I can verify that both images in reality align.
[image]

Expected behavior
The previews match in size and position.
By scaling the png down a bit, I get a correct looking preview, but in reality the png is now to small.
[image]

## 🌍  Your Environment

**Platform:**
<pre><code>
- Operating System: Ubuntu 20.04
- Luban Version:  4.1.3
</code></pre>

jane-rose, 2022-02-09:
Could u send us the png and svg file?

F1nnM, 2022-02-27:
[image]
[image]