Snapmaker/Luban #1436: Bug: Bottom layer height setting does nothing in 3d printing

HorstBaerbel, 2022-02-05:
The Bottom layer height setting does not change number of solid bottom layers produced.

### Affected Version(s)
The issue is caused by version 4.1.4

### Is this a regression? (optional)
Have not found existing issues

### To Reproduce
Steps to reproduce the behavior:
- Set bottom layer height.
- Check number of layers until infill starts
-> Bottom layer height setting does not change infill start

[image]
See above. The infill always starts at layer No. 6, no matter what the bottom layer height is.

Expected behavior
Setting a layer height of 0.4 and a bottom layer height of 0.8 should produce 2 solid layers.

## 🌍  Your Environment

**Platform:**
- Operating System: Ubuntu 20.04
- Printer: Snapmaker 2.0 A350T

jane-rose, 2022-03-07:
This is not a bug, since there is a parameter called `initial_bottom_layers` that is not exposed. We will accelerate the progress of other parameter exposure.

HorstBaerbel, 2022-03-07:
Thanks for your answer, but I don't understand. Usually in slicers you can set:
- a fixed number of bottom/top layers
- a minimum bottom/top layer height (aka "Minimum shell thickness" in PrusaSlicer)

"Top Thickness" and "Bottom Thickness" seem to be something else entirely, but then they don't work as described:
[image]

jane-rose, 2022-03-08:
> Thanks for your answer, but I don't understand. Usually in slicers you can set:
>
> * a fixed number of bottom/top layers
> * a minimum bottom/top layer height (aka "Minimum shell thickness" in [...]

Indeed this description is a bit unreasonable, but it is the default description of cura. One more thing, you can maunally change this parameter in this below file. But opening this parameter is in the plan of V4.3.
`
mac:  /Users/{username}/Library/Application Support/snapmaker-luban/Config/printing/fdmprinter.def.json
window:   %appdata%/snapmaker-luban/Config/printing/fdmprinter.def.json 
`

hollt693, 2022-08-22:
> One more thing, you can maunally change this parameter in this below file. But opening this parameter is in the plan of V4.3.
> 
> `window: %appdata%/snapmaker-luban/Config/printing/fdmprinter.def.json`

How do I change it? I opened the .json file in notepad and changed the "default_parameter" for "initial_bottom_layers" from 6 to 3, and after checking it again just now, that seems to have kept, but I don't see it reflected in the behavior of Luban. Do I need to tell it to re-read the file somehow? I apologize for being so unfamiliar with all of this.

closed 2023-03-21 by sumibin