Snapmaker/Luban #1458: Bug: xxxx in 3d printing

rkoshak, 2022-02-16:
# 🐞 bug report

### Affected Version(s)

 The issue is caused by any version after 4.0.4-beta3

### Is this a regression? (optional)

 Yes, the previous version in which this bug was not present was: 4.0.4-beta3

### To Reproduce
Steps to reproduce the behavior:
1. Start a 3D printer project
2. Import one or more STLs
3. Click on "Generate G-code"
4. Slicing happens almost instantly and it gets stopped at "Previewing G-code" at 0.5% or 1%, never progressing to completion.
 A clear and concise description of the problem...

Exception or Error (optional)
<pre><code>
None apparent.

</code></pre>

Expected behavior

I expect that the STL gets sliced and the G-code preview is generated. From there I expect to be able to export the gcode file or push it to the Snapmaker over wifi.

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: Debian Buster running on the Linux Development Environment on a Pixel Slate Chromebook
- Printer: Snapmaker A350T

</code></pre>

jane-rose, 2022-03-07:
Since we can't reproduce it. Could u send us the project file to help us reproduce it ?

rkoshak, 2022-03-08:
Here is a screen shot showing it stuck at 0.5% for Previewing G-code.

[image]

https://drive.google.com/file/d/1SNU5ZVCMcuvb4kR76oIePF4DJcbudXFE/view?usp=sharing

I put a broken project on my google drive.

All I did was upgrade to 4.1.4, created a new 3D project, added a random STL which I've successfully printed before and clicked "Generate G-code".

davodavodavo3, 2022-04-07:
I have the same issue with custome settings and materials. Stuck on 13.9%

davodavodavo3, 2022-04-07:
[Z_Offset_Test.zip](https://github.com/Snapmaker/Luban/files/8439153/Z_Offset_Test.zip)
I have attached the test STL file