Snapmaker/Luban #1473: Luban does not start on Windows 11 running on a Lenovo laptop.

srqdevg, 2022-02-21:
I downloaded Luban 4.1.4 and installed it on my new laptop which is a Lenovo Legion 7. 11th Gen i7, 32gb of RAM, 1tb SSD, NVidia GForce RTX3070. So more than sufficient for this software. When I try to run the software nothing happens, no errors dialogs, no logs that I can see. Nothing in the event logs. Any ideas as to where to look to start to determine the cause of the issue.

tyeth, 2022-02-22:
It's running fine on my Lenovo Thinkpad X1.
Maybe try one of the other releases,
https://github.com/Snapmaker/Luban/releases
or ensuring there is not a firewall/antivirus blocking Luban starting properly...

srqdevg, 2022-02-22:
I tried the latest beta release 4.2.1 beta 1 and that works. So that works for me.

closed 2022-02-22 by srqdevg