Snapmaker/Luban #1481: Bug: MAJOR - files no longer work via Touchscreen in 3d printing

tyeth, 2022-02-27:
# 🐞 bug report

### Affected Version(s)
beta 4.2.1

### Is this a regression? (optional)
yes

### To Reproduce
generate gcode, send to device over wifi, disconnect machine and browse to files, preview shows Hot End temp = 0.

Exception or Error (optional)
Printer starts printing with unheated hotend.

Expected behavior
Temperature to be read from comments where comments includes two extruders or preferably just use first M104/109 in file.

## 🌍  Your Environment

**Platform:**
- Operating System: [Windows 11 x64]
- Printer: [Snapmaker A350]

See forum post (behind private group for beta) https://forum.snapmaker.com/t/3dp-bug-running-via-touchscreen-on-machine-with-controller-firmware-4-3-2/24571

tyeth, 2022-02-27:
PDF copy of private thread attached, along with model + project file + gcode files listed in pdf/thread.

[3dp bug_ running via touchscreen on machine with controller firmware 4.3.2 - Luban Beta Program - Snapmaker_ where creation happens.pdf](https://github.com/Snapmaker/Luban/files/8148974/3dp.bug_.running.via.touchscreen.on.machine.with.controller.firmware.4.3.2.-.Luban.Beta.Program.-.Snapmaker_.where.creation.happens.pdf)
[newluban-9gservo.snap3dp.zip](https://github.com/Snapmaker/Luban/files/8148975/newluban-9gservo.snap3dp.zip)

tyeth, 2022-03-06:
They said they've made the devs aware and are working on it

vibutway, 2022-03-30:
This issue has been fixed for the latest v4.2.2 release

closed 2022-05-19 by tyeth