Snapmaker/Luban #1491: Bug: duplicate and auto arrange object in 3d printing leads to offsets and odd results

halr9000, 2022-03-03:
# 🐞 bug report

### Affected Version(s)

 The issue is caused by version 4.1.4

### To Reproduce
Steps to reproduce the behavior:
1. Create a new blank 3dp project. 
2. Choose a model such as [this one](https://1drv.ms/u/s!AvWKJeY1kZlYgqJ1OV_H6BKmy1aIrA?e=gCu1qI) which is small and can be duplicated many times. 
3. Duplicate the object until new objects overflow (they will appear red)
4. See something like the below screenshot. The objects appear slightly offset, but only on some parts of the workspace, and much space is wasted on the bed
[image]

Expected behavior

Duplication should fill the bed with the most efficient and regular fashion. A grid may be the simplest option and is what I expected, but regular would be an improvement over this irregular spacing.

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: Windows 11x64
- Printer: Snapmaker A350

</code></pre>

halr9000, 2022-03-03:
Here is a similar operation using the same model, but with Cura 4.13.1. Everything is packed tightly with no wasted space. Also I like their Multiply feature while I'm thinking about it. :) 
[image]

parachvte, 2022-03-03:
Try [v4.2.1-beta.1](https://github.com/Snapmaker/Luban/releases/tag/v4.2.1-beta.1), Auto Arrange has been optimized.

closed 2022-03-24 by sumibin