Snapmaker/Luban #1495: Bug: xxxx in 3d printing

atherr, 2022-03-04:
# 🐞 bug report

### Affected Version(s)

 The issue is caused by version 4.2.1

### Is this a regression? (optional)

 Yes, the previous version in which this bug was not present was: 3.15.2

### To Reproduce
Steps to reproduce the behavior:
1. Go to Workspace on the Luban console
2. Click on G-code file and press Start on Luban
3. At this point the Console screen disappears

Exception or Error (optional)
<pre><code>

</code></pre>

Expected behavior

I would like to be able to query the printer using console commands while the printer is actively generating a print.
This was possible in previous versions, so it is a regression.
I understand the concern of errant commands being run, but in that case auto minimize instead of removing access to the console itself. 

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: Windows 10 x64
- Printer: Snapmaker A350

</code></pre>

jane-rose, 2022-03-07:
For safely, we have hidden the console when the machine is Wifi connected in recent versions. We'll discuss it. Any progress we will update in this issue.

pauserratgutierrez, 2022-03-07:
> For safely, we have hidden the console when the machine is Wifi connected in recent versions. We'll discuss it. Any progress we will update in this issue.

I absolutelly understand the need of removing console. But it's such an usefull thing for advanced users. I suggest to deactivate it by default, but add a hidden button or any option to toggle "Advanced Mode", which shows the console and maybe more advanced things.

jane-rose, 2022-03-07:
OK. After the v4.2.2 version is ready, we will temporarily make a package for you to reopen the console . Tomorrow or the day after tomorrow. Leave a email ?

pauserratgutierrez, 2022-03-07:
> OK. After the v4.2.2 version is ready, we will temporarily make a package for you to reopen the console . Tomorrow or the day after tomorrow. Leave a email ?

Thanks for your fast reply and your patience and for wanting to create a custom package just for us to.

Will this patch be a beta test to see how will the console look in an advanced mode or only a rapidly fix to continue having it only like a provisional think? It it's that second provisional case, I appreciate it, but personally I would prefer you to confirm that a feature release will include the console in an advanced mode or something like this rather than a temporary fix just to let the time go.

Thanks in advance,
Pau

jane-rose, 2022-03-08:
> > OK. After the v4.2.2 version is ready, we will temporarily make a package for you to reopen the console . Tomorrow or the day after tomorrow. Leave a email ?
> 
> Thanks for your fast reply and your patience and for wanting to create a custom package just for us.
> 
> Will this custom package patch be a beta test for us to see how will the console look in an advanced mode or only a rapidly fix to continue having it only like a provisional thing? If it's that second provisional case, I appreciate it, but personally I would prefer you to confirm that a feature release will include the console in an advanced mode or something like this rather than a temporary fix just to let the time go.
> 
> Thanks in advance, Pau

Yeap, our current idea is to add an option in the menu whether to open the `console` when printing. Maybe in the 4.3 version.