Snapmaker/Luban #1507: Bug: xxxx Translation to German wrong

FrankBund, 2022-03-11:
Hello, 
when exiting or switching to another part of the program, I am asked how to proceed with the current project.
- Save (1)
- Cancel (2)
- Do not save (3)
However, in the German version the translation is not correct.

- Speichern (1)
- Abbrechen (2)
- Sparen Sie nicht (3)

To 3 would be here "nicht Speichern" correctly.
"Sparen Sie nicht" would be e.g. a deposit to an account.

Bug is in Version 4.2.2 and 4.2.1
[image]

vibutway, 2022-03-29:
Thank you for your feedback, this issue will be fixed in a later version

parachvte, 2023-03-21:
Has been replaced.

closed 2023-03-21 by parachvte