Snapmaker/Luban #1515: Bug: Preview and GCode wrong in laser

HorstBaerbel, 2022-03-15:
# 🐞 bug report

### Affected Version(s)
4.2.2

### Is this a regression? (optional)
Don't know

### To Reproduce
Steps to reproduce the behavior:
- Add 1 text element and change its text
- Make 2 copies
- Assign 3 different tool paths to the 3 text elements
- Go to Preview

Exception or Error (optional)
The preview has a different zoom factor and shows different texts. This is also exported to GCode and shows in the workspace.
Editing the copied text makes that text correct appear in preview and workspace.
[image]
[image]

Expected behavior
Copied elements should behave correctly and export correct gcode

## 🌍  Your Environment

**Platform:**
- Operating System: Ubuntu 20.04
- Printer: Snapmaker A350T

yufang-123, 2022-03-29:
Thank you for your feedback, we have confirmed that the problem exists and will fix it in subsequent versions

HorstBaerbel, 2022-04-20:
Seems to be fixed in 4.2.3