Snapmaker/Luban #1517: Bug: Bad Nozzle-Temp in 3d printing Luban 4.2.2

ThM-58, 2022-03-17:
Hello, everyone
when generating the G code, the nozzle temperature 
is entered incorrectly! The correct temperature is 
in the header (245°), but 215° is generated in the actual G-code!
(M104 S215; M109 S215)

<pre><code>
;file_total_lines: 55533
;estimated_time(s): 9649.26
;nozzle_temperature(°C): 245
;nozzle_1_temperature(°C): null
;nozzle_0_diameter(mm): 0.4
;nozzle_1_diameter(mm): null
;build_plate_temperature(°C): 80
;work_speed(mm/minute): 3000
;max_x(mm): 10.5
;max_y(mm): 15.949999809265137
;max_z(mm): 29.299999237060547
;min_x(mm): -10.5
;min_y(mm): -15.949999809265137
;min_z(mm): 0

;Header End

; G-code for 3dp engraving
; Generated by Snapmaker Luban 4.2.2
; Thu Mar 17 2022 17:30:13

T0
M82 ;absolute extrusion mode
;Start GCode begin
M104 S215
M140 S80
G28 ;home
G90 ;absolute positioning
G1 X-10 Y-10 F3000
G1 Z0 F1800
M109 S215;Wait for Hotend Temperature
M190 S80;Wait for Bed Temperature
</code></pre>

**Platform:**
OS: macOS Big Sur
Printer: Snapmaker 2.0 A350
Luban 4.2.2

Best regards
ThM

yufang-123, 2022-03-29:
This issue is difficult to reproduce, we need you to provide a more specific description of the steps or documents, so that we can reproduce the problem

davodavodavo3, 2022-04-07:
I have the same issues with custom material

davodavodavo3, 2022-04-07:
Steps to reproduce the behavior:
1. Create new material inside PETG profile. For example petg E-Sun
2. Set the nozzle temperature to 245
3. Close Luban.
4. Open Luban again. The previosuly selected profile will be active.
5. Generate G-Code

Inside G-code the nozzle temp will be 215

davodavodavo3, 2022-04-07:
[image]
[image]

Other settings seems correct for the custom material exept nozzle temp

PatonLewis, 2023-01-08:
Still seeing this bug almost a year later, and there's no effective work-around. The nozzle and bed temperatures are often not the ones specified in Luban. Specify a bed temp of 70C, it uses a bed temp of 75C. Specify a nozzle temp of 230C, it uses a nozzle temp of 235C. This makes it significantly harder to debug and fix printing problems with new filaments.

pmzandbergen, 2023-02-26:
Confirmed the same issue in version 4.6.2

parachvte, 2023-02-27:
> Confirmed the same issue in version 4.6.2

Should be fixed in 4.6.3

pmzandbergen, 2023-02-27:
> > Confirmed the same issue in version 4.6.2
> 
> Should be fixed in 4.6.3

Unfortunately, just reconfirmed in 4.6.3

closed 2023-03-21 by sumibin