Snapmaker/Luban #1518: New project, directory issue

Obispud, 2022-03-17:
Since updating to Luban 4.1.4 (the latest version) then starting a new 3D printing project, the directory you want to load an .stl from now defaults to the same directory path containing a previous printed files.

Let me explain: Luban always opens this directory for me 'I:\3D Prints\Painting\Paint rack' and shows 'custom files'! In the past I could select a directory from 'This PC' and 'all files' would be shown making it easier to navigate to the directory and .stl I want to print. This gets rather annoying having to clear the 'custom files' setting, chose the directory and files with the .stl you want to print.

I see no way to change that default setting within Luban.

yufang-123, 2022-03-29:
Thank you for your feedback, we have confirmed the existence of the problem and have now fed back to the development process.

JThobbyist, 2022-11-23:
What's the status of this? This seems like it would be a quick fix, and it definitely irks me to have to change folders every time. Why not make it have the same behavior as `File > Import Object`? Is it not meant to be a shortcut for that same exact function anyways?