Snapmaker/Luban #1523: Bug: Simplify3D gcode Settings Commented Out

jenkins007, 2022-03-25:
# 🐞 bug report

### Affected Version(s)

 Luban 4.2.2

### Is this a regression? (optional)

 unknown: ....

### To Reproduce
Steps to reproduce the behavior:
1. Open .stl file in Simplify3D.
2. Edit process settings.
3. Prepare to print.
4. Save to disk or USB.
5. Open the gcode with text edit (on Mac).
6. Note the entire top section where the simplify3d settings are generated are commented out.
 A clear and concise description of the problem...

Exception or Error (optional)
<pre><code>

jenkins007, 2022-03-25:
[Simplify3D Settings Ignored.gcode.txt](https://github.com/Snapmaker/Luban/files/8346921/Simplify3D.Settings.Ignored.gcode.txt)

sumibin, 2022-03-25:
I'm not quite sure how this issue is related to Luban. Can you describe what your actual problem is?

jenkins007, 2022-03-25:
You are correct. The output from Simplify3D has these settings commented out. I will notify the appropriate channel for Simplify3d to get this addressed and close this issue.

closed 2022-03-25 by jenkins007