Snapmaker/Luban #1558: Bug: xxxx in 3d printing

davodavodavo3, 2022-04-20:
# 🐞 bug report

### Affected Version(s)

 The issue is caused by version 4.2.3

### To Reproduce
Steps to reproduce the behavior:
1. Started the printing in Luban

Exception or Error
Build Plate target Temperature is not showing

Expected behavior
Build Plate Temperature should show the target temp.

## 🌍  Your Environment

**Platform:**
Windows 11 x64
Printer: Snapmaker A350T

[image]

polrus, 2022-11-28:
the bug still exists in 4.4 + nozzle  temp does not show either
tested in Opensuse Linux
workaround is to disconnect and reconnect to the printer

sumibin, 2023-03-21:
This issue may have been fixed in 4.7. If it still exists, please reopen it.