Snapmaker/Luban #1577: Bug: luban stopped working when I upgraded to Ubuntu/Pop OS 22.04 LTS

arunbatchu, 2022-04-26:
Hi, none of the latest ( I tried through January 2022 ) .deb distros seem to work on the latest 22.04 LTS  PopOS  ( Ubuntu )  OS. Installs OK, but does not run.  What can I do to provide more info?

arunbatchu, 2022-04-27:
2022-04-26T23:59:56.509Z - info init Starting the server at http://127.0.1.1:42795
[6290:0426/185956.593748:FATAL:gpu_data_manager_impl_private.cc(445)] GPU process isn't usable. Goodbye.
Trace/breakpoint trap (core dumped)

Could this be related to the Wayland vs. x.org choices when launching 22.04 ?

HorstBaerbel, 2022-06-04:
This seems to be an [electron-related issue](https://github.com/atom/atom/issues/23608). A workaround is starting snapmaker-luban with `--in-process-gpu` and/or `--no-sandbox`.

guillaumedavidphd, 2022-06-04:
`--in-process-gpu` made it work for me, waiting for a more permanent fix. Thanks!

sniper614, 2022-08-19:
I am having the same issue when I upgraded to Mint 21

particleflux, 2022-09-06:
For me, `--disable-gpu-sandbox` worked too.

Each of those three work around the issue:

* `--disable-gpu-sandbox`
* `--no-sandbox`
* `--in-process-gpu`

HorstBaerbel, 2022-09-16:
This is still an issue with 4.3.2. Please fix this, it is very inconvenient! There are multiple solutions and the core reason mentioned here.

HorstBaerbel, 2022-09-18:
A workaround is adding the `--disable-gpu-sandbox` option to /opt/Snapmaker Luban/resources/app/main.js after `var _electron = require("electron");`
```javascript
var _electron = require("electron");
_electron.app.commandLine.appendSwitch("--disable-gpu-sandbox");
```

HorstBaerbel, 2022-11-30:
Still not fixed in 4.4.0