Snapmaker/Luban #1585: Bug: Luban stopped working for Laser Cutting after Manual Origin

Galindorf, 2022-05-01:
# 🐞 bug report

### Affected Version(s)

 The issue is caused by version 1.4.3

### Is this a regression? (optional)

 Yes, the previous version in which this bug was not present was: this same version, yesterday

### To Reproduce
Steps to reproduce the behavior:
After running files via  USB and WiFi sent to the machine and manually setting work origin: 
Open 3 axis laser environment
load a file for cutting and set your settings
export GCode to Workspace
Start Job via Luban
GCode loads to machine successfully, machine begins to move toolhead and then... stops. No error on Luban. Behaves as though file is finished, with no time lapse shown.  
 A clear and concise description of the problem...
Earlier today I was running experiments to learn how to manually set the work origin on my projects. I successfully ran cut files loaded both via USB and WiFi file transfer. However, when I attempted to run a cut file from Luban, with or without the Camera Capture, it does not work. No error, it just... stops. 
Possibly related: my machine is not staying connected very well at all, and when I reconnect it pops up the screen authorization request. Sometimes it disappears and connects, other times I have to get up and use the touch screen to authorize. My computer and machine are both on the same WiFi network, also I have the "auto connect" checkbox filled in. 

Exception or Error (optional)
<pre><code>

</code></pre>

Expected behavior

I expect when I click "Start Job on Luban" with Auto Mode, it will actually start the file and cut. 

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: Windows10 x64
- Printer: Snapmaker A350 with enclosure, 10w laser module
</code></pre>

Galindorf, 2022-05-01:
More Info: 
I tried to send the file that would not cut via Luban to the machine. The preview loaded (simple circle cut for test), the auto thickness and focus worked via touchscreen, I set a manual work origin. When I touched "Start" I got an error message that read "Unable to Start Job, please check your machine settings and try again"

I turned the machine off and back on, deleted all the files from the touch screen, restarted Luban. When I used the Camera Capture, it took only this: 
[image]
I exited to find my machine had disconnected again. I reconnected, deleted the background, tried a second time with camera capture and it worked properly this time. I again drew on a circle, set to standard Cutting 200G Cardstock preset. I loaded to Workspace and started via Luban and it worked this time... (although the settings did not cut). By the time I checked back on Luban it was disconnected again. 

If it helps, here is the Console:

snapmaker-luban 4.2.3
Connected via Wi-Fi
G54
Select workspace 0
X:-60.00 Y:-31.00 Z:81.80 E:0.00 Count X:52400 Y:56000 Z:60000 B:0
ok
G53
ok
G28
X:-19.00 Y:347.00 Z:334.00 E:0.00 Count X:0 Y:142800 Z:133600 B:0
ok
G54
Select workspace 0
X:-191.00 Y:186.00 Z:265.80 E:0.00 Count X:0 Y:142800 Z:133600 B:0
ok
snapmaker-luban 4.2.3
Connected via Wi-Fi
G54
Select workspace 0
X:60.00 Y:17.00 Z:221.80 E:0.00 Count X:100400 Y:75200 Z:116000 B:0
ok
snapmaker-luban 4.2.3
Connected via Wi-Fi
G54
ok
snapmaker-luban 4.2.3
Connected via Wi-Fi
G54
ok
snapmaker-luban 4.2.3
Connected via Wi-Fi
G54
ok

Galindorf, 2022-05-02:
Trying again today to cut from Luban. Created a square to vector cut from within Luban, sent to Workspace, Start on Luban (manual mode, set material thickness myself).  GCode successfully sent. Toolhead went to proper spot to begin, then immediately homed itself as though the job was done.  Job status shown here: 
[image]

Console readout:
------------------------------------
snapmaker-luban 4.2.3
Connected via Wi-Fi
G54
Select workspace 0
X:112.00 Y:130.00 Z:121.33 E:0.00 Count X:52400 Y:56000 Z:60000 B:0
ok
G53
ok
G28
X:-19.00 Y:347.00 Z:334.00 E:0.00 Count X:0 Y:142800 Z:133600 B:0
ok
G54
Select workspace 0
X:-19.00 Y:347.00 Z:305.33 E:0.00 Count X:0 Y:142800 Z:133600 B:0
ok

Galindorf, 2022-05-02:
GCode Below:
__________________________________________________
;Header Start
;header_type: laser
;renderMethod: line
;file_total_lines: 44
;estimated_time(s): 12.821
;is_rotate: false
;diameter: 40
;max_x(mm): -125.102
;max_y(mm): -137.945
;max_z(mm): 0
;max_b(mm): 0
;min_x(mm): -154.26000000000002
;min_y(mm): -167.851
;min_b(mm): 0
;min_z(mm): 0
;work_speed(mm/minute): 600
;jog_speed(mm/minute): 3000
;power(%): 100
;thumbnail: data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAtAAAAHgCAYAAACMxVqsAAAgAElEQVR4Xu3WsW2cZxBF0VUlTA21JFdCKjRVidWS4JRFCA5t0IBzTcCX3LPxhx3M+Se4n37+/Pufhx8BAgQIECBAgAABAr8k8ElA/5KTRwQIECBAgAABAgT+ExDQDoEAAQIECBAgQIDAQUBAH7A8JUCAAAECBAgQICCg3QABAgQIECBAgACBg4CAPmB5SoAAAQIECBAgQEBAuwECBAgQIECAAAECBwEBfcDylAABAgQIECBAgICAdgMECBAgQIAAAQIEDgIC+oDlKQECBAgQIECAAAEB7QYIECBAgAABAgQIHAQE9AHLUwIECBAgQIAAAQIC2g0QIECAAAECBAgQOAgI6AOWpwQIECBAgAABAgQEtBsgQIAAAQIECBAgcBAQ0AcsTwkQIECAAAECBAgIaDdAgAABAgQIECBA4CAgoA9YnhIgQIAAAQIECBAQ0G6AAAECBAgQIECAwEFAQB+wPCVAgAABAgQIECAgoN0AAQIECBAgQIAAgYOAgD5geUqAAAECBAgQIEBAQLsBAgQIECBAgAABAgcBAX3A8pQAAQIECBAgQICAgHYDBAgQIECAAAECBA4CAvqA5SkBAgQIECBAgAABAe0GCBAgQIAAAQIECBwEBPQBy1MCBAgQIECAAAECAtoNECBAgAABAgQIEDgICOgDlqcECBAgQIAAAQIEBLQbIECAAAECBAgQIHAQENAHLE8JECBAgAABAgQICGg3QIAAAQIECBAgQOAgIKAPWJ4SIECAAAECBAgQENBugAABAgQIECBAgMBBQEAfsDwlQIAAAQIECBAgIKDdAAECBAgQIECAAIGDgIA+YHlKgAABAgQIECBAQEC7AQIECBAgQIAAAQIHAQF9wPKUAAECBAgQIECAgIB2AwQIECBAgAABAgQOAgL6gOUpAQIECBAgQIAAAQHtBggQIECAAAECBAgcBAT0ActTAgQIECBAgAABAgLaDRAgQIAAAQIECBA4CAjoA5anBAgQIECAAAECBAS0GyBAgAABAgQIECBwEBDQByxPCRAgQIAAAQIECAhoN0CAAAECBAgQIEDgICCgD1ieEiBAgAABAgQIEBDQboAAAQIECBAgQIDAQUBAH7A8JUCAAAECBAgQICCg3QABAgQIECBAgACBg4CAPmB5SoAAAQIECBAgQEBAuwECBAgQIECAAAECBwEBfcDylAABAgQIECBAgICAdgMECBAgQIAAAQIEDgIC+oDlKQECBAgQIECAAAEB7QYIECBAgAABAgQIHAQE9AHLUwIECBAgQIAAAQIC2g0QIECAAAECBAgQOAgI6AOWpwQIECBAgAABAgQEtBsgQIAAAQIECBAgcBAQ0AcsTwkQIECAAAECBAgIaDdAgAABAgQIECBA4CAgoA9YnhIgQIAAAQIECBAQ0G6AAAECBAgQIECAwEFAQB+wPCVAgAABAgQIECAgoN0AAQIECBAgQIAAgYOAgD5geUqAAAECBAgQIEBAQLsBAgQIECBAgAABAgcBAX3A8pQAAQIECBAgQICAgHYDBAgQIECAAAECBA4CAvqA5SkBAgQIECBAgAABAe0GCBAgQIAAAQIECBwEBPQBy1MCBAgQIECAAAECAtoNECBAgAABAgQIEDgICOgDlqcECBAgQIAAAQIEBLQbIECAAAECBAgQIHAQENAHLE8JECBAgAABAgQICGg3QIAAAQIECBAgQOAgIKAPWJ4SIECAAAECBAgQENBugAABAgQIECBAgMBBQEAfsDwlQIAAAQIECBAgIKDdAAECBAgQIECAAIGDgIA+YHlKgAABAgQIECBAQEC7AQIECBAgQIAAAQIHAQF9wPKUAAECBAgQIECAgIB2AwQIECBAgAABAgQOAgL6gOUpAQIECBAgQIAAAQHtBggQIECAAAECBAgcBAT0ActTAgQIECBAgAABAgLaDRAgQIAAAQIECBA4CAjoA5anBAgQIECAAAECBAS0GyBAgAABAgQIECBwEBDQByxPCRAgQIAAAQIECAhoN0CAAAECBAgQIEDgICCgD1ieEiBAgAABAgQIEBDQboAAAQIECBAgQIDAQUBAH7A8JUCAAAECBAgQICCg3QABAgQIECBAgACBg4CAPmB5SoAAAQIECBAgQEBAuwECBAgQIECAAAECBwEBfcDylAABAgQIECBAgICAdgMECBAgQIAAAQIEDgIC+oDlKQECBAgQIECAAAEB7QYIECBAgAABAgQIHAQE9AHLUwIECBAgQIAAAQIC2g0QIECAAAECBAgQOAgI6AOWpwQIECBAgAABAgQEtBsgQIAAAQIECBAgcBAQ0AcsTwkQIECAAAECBAgIaDdAgAABAgQIECBA4CAgoA9YnhIgQIAAAQIECBAQ0G6AAAECBAgQIECAwEFAQB+wPCVAgAABAgQIECAgoN0AAQIECBAgQIAAgYOAgD5geUqAAAECBAgQIEBAQLsBAgQIECBAgAABAgcBAX3A8pQAAQIECBAgQICAgHYDBAgQIECAAAECBA4CAvqA5SkBAgQIECBAgAABAe0GCBAgQIAAAQIECBwEBPQBy1MCBAgQIECAAAECAtoNECBAgAABAgQIEDgICOgDlqcECBAgQIAAAQIEBLQbIECAAAECBAgQIHAQENAHLE8JECBAgAABAgQICGg3QIAAAQIECBAgQOAgIKAPWJ4SIECAAAECBAgQENBugAABAgQIECBAgMBBQEAfsDwlQIAAAQIECBAgIKDdAAECBAgQIECAAIGDgIA+YHlKgAABAgQIECBAQEC7AQIECBAgQIAAAQIHAQF9wPKUAAECBAgQIECAgIB2AwQIECBAgAABAgQOAgL6gOUpAQIECBAgQIAAAQHtBggQIECAAAECBAgcBAT0ActTAgQIECBAgAABAgLaDRAgQIAAAQIECBA4CAjoA5anBAgQIECAAAECBAS0GyBAgAABAgQIECBwEBDQByxPCRAgQIAAAQIECAhoN0CAAAECBAgQIEDgICCgD1ieEiBAgAABAgQIEBDQboAAAQIECBAgQIDAQUBAH7A8JUCAAAECBAgQICCg3QABAgQIECBAgACBg4CAPmB5SoAAAQIECBAgQEBAuwECBAgQIECAAAECBwEBfcDylAABAgQIECBAgICAdgMECBAgQIAAAQIEDgIC+oDlKQECBAgQIECAAAEB7QYIECBAgAABAgQIHAQE9AHLUwIECBAgQIAAAQIC2g0QIECAAAECBAgQOAgI6AOWpwQIECBAgAABAgQEtBsgQIAAAQIECBAgcBAQ0AcsTwkQIECAAAECBAgIaDdAgMBE4PX1j8kcQ9oCLy9f2wC2J0BgIiCgJ8yGEGgLvMfz9+9/thFs/+ECb29vj+fnl4eI/nBqAwjkBQR0/gQAEPh4gf8D+sePvz5+mAlZgc+ff3t8+fK7gM5egMUJ7AQE9M7aJAJZAQGd/fTTxQX0lNswAmkBAZ3+/JYnsBEQ0Bvn+hQBXb8A+xPYCQjonbVJBLICAjr76aeLC+gpt2EE0gICOv35LU9gIyCgN871KQK6fgH2J7ATENA7a5MIZAUEdPbTTxcX0FNuwwikBQR0+vNbnsBGQEBvnOtTBHT9AuxPYCcgoHfWJhHICgjo7KefLi6gp9yGEUgLCOj057c8gY2AgN4416cI6PoF2J/ATkBA76xNIpAVENDZTz9dXEBPuQ0jkBYQ0OnPb3kCGwEBvXGuTxHQ9QuwP4GdgIDeWZtEICsgoLOffrq4gJ5yG0YgLSCg05/f8gQ2AgJ641yfIqDrF2B/AjsBAb2zNolAVkBAZz/9dHEBPeU2jEBaQECnP7/lCWwEBPTGuT5FQNcvwP4EdgICemdtEoGsgIDOfvrp4gJ6ym0YgbSAgE5/fssT2AgI6I1zfYqArl+A/QnsBAT0ztokAlkBAZ399NPFBfSU2zACaQEBnf78liewERDQG+f6FAFdvwD7E9gJCOidtUkEsgICOvvpp4sL6Cm3YQTSAgI6/fktT2AjIKA3zvUpArp+AfYnsBMQ0DtrkwhkBQR09tNPFxfQU27DCKQFBHT681uewEZAQG+c61MEdP0C7E9gJyCgd9YmEcgKCOjsp58uLqCn3IYRSAsI6PTntzyBjYCA3jjXpwjo+gXYn8BOQEDvrE0ikBUQ0NlPP11cQE+5DSOQFhDQ6c9veQIbAQG9ca5PEdD1C7A/gZ2AgN5Zm0QgKyCgs59+uriAnnIbRiAtIKDTn9/yBDYCAnrjXJ8ioOsXYH8COwEBvbM2iUBWQEBnP/10cQE95TaMQFpAQKc/v+UJbAQE9Ma5PkVA1y/A/gR2AgJ6Z20SgayAgM5++uniAnrKbRiBtICATn9+yxPYCAjojXN9ioCuX4D9CewEBPTO2iQCWQEBnf3008UF9JTbMAJpAQGd/vyWJ7ARENAb5/oUAV2/APsT2AkI6J21SQSyAgI6++mniwvoKbdhBNICAjr9+S1PYCMgoDfO9SkCun4B9iewExDQO2uTCGQFBHT2008XF9BTbsMIpAUEdPrzW57ARkBAb5zrUwR0/QLsT2AnIKB31iYRyAoI6Oynny4uoKfchhFICwjo9Oe3PIGNgIDeONenCOj6BdifwE5AQO+sTSKQFRDQ2U8/XVxAT7kNI5AWENDpz295AhsBAb1xrk8R0PULsD+BnYCA3lmbRCArIKCzn366uICechtGIC0goNOf3/IENgICeuNcnyKg6xdgfwI7AQG9szaJQFZAQGc//XRxAT3lNoxAWkBApz+/5QlsBAT0xrk+RUDXL8D+BHYCAnpnbRKBrICAzn766eICesptGIG0gIBOf37LE9gICOiNc32KgK5fgP0J7AQE9M7aJAJZAQGd/fTTxQX0lNswAmkBAZ3+/JYnsBEQ0Bvn+hQBXb8A+xPYCQjonbVJBLICAjr76aeLC+gpt2EE0gICOv35LU9gIyCgN871KQK6fgH2J7ATENA7a5MIZAUEdPbTTxcX0FNuwwikBQR0+vNbnsBGQ [... cut ...] ECAAAECBA4CAvqA5SkBAgQIECBAgAABAe0GCBAgQIAAAQIECBwEBPQBy1MCBAgQIECAAAECAtoNECBAgAABAgQIEDgICOgDlqcECBAgQIAAAQIE/gWAvNqVX3KlSAAAAABJRU5ErkJggg==
;Header End

; Laser multi-pass, pass 1 with Z = 0
; G-code for laser engraving
; Generated by Snapmaker Luban
; G-code START <<<
G90
G21
M106 P0 S255
G0 F3000
G1 F600
G0 X-154.26 Y-137.94
M3 P100 S255
M3
G1 X-125.10 Y-137.94
G1 X-125.10 Y-167.85
G1 X-154.26 Y-167.85
G1 X-154.26 Y-137.94
M5
G0 X-139.68 Y-152.90
M107 P0
; G-code END <<<
G91
G0 Z0 F150
G90

Galindorf, 2022-05-02:
More notes: 
I uninstalled and reinstalled Luban. So far I have run 4 successful files through Luban. I am hoping this resolved the issue, will update if needed. Just wanted to let others know!

sumibin, 2023-03-21:
This problem, if it still occurs in the new version, please reopen it.