Snapmaker/Luban #1587: Feature Request: Enter Material Thickness during manual Work Origin

Galindorf, 2022-05-04:
# 🐞 feature request

### Is your feature request related to a problem? Please describe.
I am working on projects where I have to set a manual work origin. If it is a dark, red or shiny material I can't use the auto focus, and have to use the manual focus. It takes forever and more often than not, displaces or moves the object I'm trying to engrave. 

### Describe the solution you'd like?

Please add the ability to dial in the material thickness via manual mode, instead of having to use the card every time. If I already know my material is 1.5mm (measured via caliper) I don't want to have to go through the whole process of moving the toolhead down... every single time I set a manual work origin. 
Thank you!

HorstBaerbel, 2022-06-27:
Is this for Laser / CNC? This should be possible now with 4.3.x

Galindorf, 2022-07-03:
Moving to Firmware

closed 2022-07-03 by Galindorf