Snapmaker/Luban #1590: Bug: Laser: Fill Line direction ignored

tyeth, 2022-05-16:
# 🐞 bug report

### Affected Version(s)
4.3.0

### Is this a regression? 
Yes, there is a bug listed against 3.15 that shows it working in diagonal mode

### To Reproduce
Steps to reproduce the behavior:
1. Add image and choose vector mode. Create engraving toolpath with Fill, Line mode, Direction:diagonal2, and click generate.
2. Look at the preview, zoom in, the lines are horizontal.

## 🌍  Your Environment

**Platform:**

- Operating System: [Windows 11 x64]
- Printer: [Snapmaker A350 + 10W Laser]

First pointed to from https://www.reddit.com/r/snapmaker/comments/umfsr9/vertical_path_for_laser/
[image]
[image]

tyeth, 2022-05-16:
[BW_sunset_line_diagonal_0398de10.zip](https://github.com/Snapmaker/Luban/files/8703360/BW_sunset_line_diagonal_0398de10.zip)

jane-rose, 2022-06-06:
It's abnormal, but I can't reproduce it, maybe you can help us to provide steps for stable reproduction.

tyeth, 2022-08-03:
No sorry. Still happening in 4.3.2
Try switching engraving material and adjusting modes more. Also I visit the workspace before coming back and adding the model / setting up the engrave.
I just did another and it's only using horizontal. I chose aluminium vector, and then switched from Line to Fill, then change direction in the settings cog.

closed 2023-03-21 by sumibin