Snapmaker/Luban #1600: Bug: old issue #640 appearing again on WiFi connect in 3d printing in just update Luban v4.3.0

duppypro, 2022-05-28:
# 🐞 bug report

### Affected Version(s)
 The issue is caused by version 4.3.0
[edit] I also did update to MacOS Monterey 12.4 between last print and this attempt.
I did not have the issue a few days ago.  Today I update first then tried to WiFi connect and failed.

### Is this a regression? (optional)
Yes, the previous version in which this bug was not present was: v4.2.3

### To Reproduce
I am seeing the same white screen when pressing Connect button after I update to Luban 4.3.0 today.
Similar to #640 

After restating the app and the printer. I no longer get a white screen but I do immediately get a "Failed to connect, Invalid authentication."

Luban Workspace is finding the printer and the IP shown in Luban matches the IP shown on the printer About menu.

Very frustrating to not be able to print until this is fixed.

Does this console error help?
-------
<pre><code>
app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2 TypeError: Cannot read properties of undefined (reading 'emergencyStopButton')
    at Ud (app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2:4874777)
    at to (app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2:2952238)
    at Bo (app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2:2961756)
    at Fs (app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2:3005648)
    at El (app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2:2991686)
    at wl (app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2:2991614)
    at _l (app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2:2991477)
    at pl (app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2:2988464)
    at app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2:2938133
    at t.unstable_runWithPriority (app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2:3015321)
as @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
n.callback @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
ca @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
ps @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
Cl @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
t.unstable_runWithPriority @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
Fi @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
kl @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
pl @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
(anonymous) @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
t.unstable_runWithPriority @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
Fi @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
Ui @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
Hi @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
fl @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
notify @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
notifyNestedSubs @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
i @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
g @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
(anonymous) @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
(anonymous) @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
dispatch @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
Marlin:state @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
(anonymous) @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
(anonymous) @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
(anonymous) @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
r.emit @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
emit @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
emitEvent @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
onevent @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
onpacket @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
r.emit @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
emitReserved @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
ondecoded @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
r.emit @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
add @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
ondata @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
r.emit @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
onPacket @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
(anonymous) @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
r.emit @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
onPacket @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
onData @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
ws.onmessage @ app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2
Show 14 more frames
app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2 Uncaught TypeError: Cannot read properties of undefined (reading 'emergencyStopButton')
    at Ud (app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2:4874777)
    at to (app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2:2952238)
    at Bo (app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2:2961756)
    at Fs (app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2:3005648)
    at El (app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2:2991686)
    at wl (app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2:2991614)
    at _l (app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2:2991477)
    at pl (app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2:2988464)
    at app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2:2938133
    at t.unstable_runWithPriority (app.d272919c74a3c43b9770.bundle.js?_=1652154662414:2:3015321)
</code></pre>

Expected behavior
Expected the Confirm connection dialog to appear on the printer and Luban to connect over WiFi to printer.

## 🌍  Your Environment

**Platform:**
<pre><code>
- Operating System:  MacOS Monterey 12.4
- Printer: [Snapmaker 2 A250]

</code></pre>

jane-rose, 2022-05-29:
Ok, we have check for this yesterday. But we don't reproduce it, we will check it again soon with the information you provided.

jane-rose, 2022-05-31:
Have to update firmware version for Snapmaker 2.0 equal or above 1.14.1 to use wifi connection. And it will be fixed in subsequent versions. Any progress can update here.