Snapmaker/Luban #1601: Bug: Slow performance in Luban 4.3

imh1kiko, 2022-05-29:
# 🐞 bug report

### Affected Version(s)
Version 4.3

### Is this a regression?
The previous version (4.2) ran just fine. Now the UI is hanging.

### To Reproduce
Steps to reproduce the behavior:

Case 1
1. Go to workspace
2. Connect
3. Printer has already connected, Luban is still hanging.

Case 2
1. Click on generate Gcode.
2. GCode generation takes longer.

Case 3
1. While printing (lasering, in my case), go back to 3d/laser/cnc tab.
2. Now go back to workspace.
3. Luban freezes, resumes after a good while.

Expected behavior
Better performance. In 4.2 latest release, everything was snappy.

## 🌍  Your Environment
- Operating System: Windows 10 x64
- Printer: Snapmaker A150

jane-rose, 2022-05-29:
Ok, we will check for this soon.

jane-rose, 2022-05-31:
We can only reproduce case 1, and it will be fixed in subsequent versions. Others are consistent with the 4.2 version or we cannot reproduce. Any futher progress or information can update here.

jane-rose, 2022-06-02:
Ok, now we can reproduce case3, and we will re-check case4. But the reason is not slow performance but reconnect action when switching page. (You can temporarily turn off the auto-connect checkbox to solve it .)All these will be fixed in subsequent versions. Thanks for your help.