Snapmaker/Luban #1610: Bug: Fill tool path mode not working in CNC (and probably laser too)

HorstBaerbel, 2022-06-05:
# 🐞 bug report

### Affected Version(s)
4.3.0

### Is this a regression? (optional)
no idea

### To Reproduce
- Add attached file
- Add toolpath and set mode to "fill"
[image]

### Exception or Error (optional)
Area not filled. Probably related to line direction, see #1609.
[image]

### Expected behavior
Area filled correctly

## 🌍  Your Environment

**Platform:**
- Operating System: Fedora 36
- Printer: Snapmaker 2.0 A350T

[Innen.zip](https://github.com/Snapmaker/Luban/files/8840165/Innen.zip)

HorstBaerbel, 2022-06-06:
How on earth is this a feature and not a bug?

jane-rose, 2022-06-06:
Ok, we can reproduce it. Sorry for that happend. But we may definite it as a new feature which can merge or distinguish line segments with intersections into closed graphics. As this time, you can draw that dxf using `spline` but not individual line segments when using `fill` mode. See file attached. 
[four.dxf.zip](https://github.com/Snapmaker/Luban/files/8843460/four.dxf.zip)

HorstBaerbel, 2023-04-15:
Error still in 4.7.2 and can be tested with the same files. Please reopen.
[image]