Snapmaker/Luban #1611: Bug: Incorrect tool path selected for selected object in CNC (and probably laser too)

HorstBaerbel, 2022-06-05:
# 🐞 bug report

### Affected Version(s)
4.3.0

### Is this a regression? (optional)
no idea

### To Reproduce
- Add 2 objects with 2 different tool paths (object A -> tool path 1, object B -> tool path 2)
- Select tool path 2
- Select object A

### Exception or Error (optional)
Tool path 2 is selected and displayed, but does not belong to selected object A.

### Expected behavior
Selecting an object should automatically select its associated tool path, otherwise you edit the wrong tool path.

## 🌍  Your Environment

**Platform:**
- Operating System: Fedora 36
- Printer: Snapmaker 2.0 A350T

sumibin, 2022-06-06:
Selecting Toolpath and selecting objects are two completely decoupled operations.
When Toolpath is selected, objects are displayed in blue.
[image]

HorstBaerbel, 2022-06-06:
> Selecting Toolpath and selecting objects are two completely decoupled operations. When Toolpath is selected, objects are displayed in blue.

I know. I wrote that (sorry for the edit). But imo it makes sense to _automatically select the tool path of an object, when selecting that object_. What do you want to do when selecting an object? You want to manipulate it or create / pick / change its tool path. What do you want do do when selecting a tool path? View / modify the tool path, but not the objects that have that tool path (there can be multiple objects).

I'd propose this:

- Selecting an object selects its tool path too and it can be changed (right tool path pane). If multiple objects are selected, multiple tool paths are highlighted, but can't be changed (right tool pane grayed out). Objects can be assigned different tool paths by selecting the object and creating a tool path, double clicking a tool path or pressing a button "apply to selected object".

- Selecting a tool path highlights objects that have that tool path (different color?) and it can be changed in the right pane. Selecting a tool path does not change object selection.

Tell me if this makes no sense or I have forgotten some details...

closed 2023-03-21 by sumibin