Snapmaker/Luban #1612: Bug: Can start CNC job with open enclosure door, though door detection is activated

HorstBaerbel, 2022-06-06:
# 🐞 bug report

### Affected Version(s)
4.3.0

### Is this a regression? (optional)
no idea

### To Reproduce
- Open an enclosure door
- Make CNC job and go to workspace
- Door detection is turned on (see also #1441)
- Start job

### Exception or Error (optional)
Job starts though enclosure door is open

### Expected behavior
Job should not be startable with an open door if "Door Detection" is activated. I haven't tried, but if this happens for laser cutting it is a big no-no.

## 🌍  Your Environment

**Platform:**
- Operating System: Fedora 36
- Printer: Snapmaker A350T

jane-rose, 2022-06-07:
Since we can't reproduce it, could u tell more information or steps to help us reproduce it. Like the firmware version, is using wifi connection?

HorstBaerbel, 2022-06-07:
Using a WiFi connection. Firmware is 1.14.2.

Sorry, I got the sequence wrong. This makes it reproducible for me:

- MACHINE MUST BE OFF
- Open an enclosure door
- TURN THE MACHINE ON
- Make CNC job and go to workspace
- Door detection is turned on in Luban
- Start job

Door detection is turned on in Luban and visible in machine enclosure settings, so the setting must be correct.

jane-rose, 2022-06-07:
Ok, we will re-check it. Btw,  i mean we can't reproduce issue 1441 neither

jane-rose, 2022-06-09:
OK, we confirm it is a **firmware** issue, any progress will be updated in this issue. Thx.