Snapmaker/Luban #1617: Bug: Copied objects not in preview / workspace in Laser

HorstBaerbel, 2022-06-08:
# 🐞 bug report

### Affected Version(s)
4.3.0

### Is this a regression? (optional)
no idea

### To Reproduce
Steps to reproduce the behavior:
1. Add objects from svg (see attached file), assign different tool paths or load snaplzr file
2. Duplicate objects (Mirror or not, does not matter)
4. Go to preview and / or workspace
[image]

### Exception or Error (optional)
Only one object is on preview and workspace
[image]

### Expected behavior
Copied object should be visible in preview and export to workspace

## 🌍  Your Environment

**Platform:**
- Operating System: Fedora 36
- Printer: Snapmaker 2.0 A350T

[Bug_again.zip](https://github.com/Snapmaker/Luban/files/8863422/Bug_again.zip)

sumibin, 2023-03-21:
This issue should have been fixed.

HorstBaerbel, 2023-04-15:
Error still in 4.7.2. Can be tested with same files. Please reopen.