Snapmaker/Luban #1618: Bug: 4.30.1 ALPHA top/shell preview in 3d printing

tyeth, 2022-06-08:
New Alpha, has preview issue, direction of shell/top line when thin wall (one outer surrounding one middle/shell/top line) see private beta post
https://forum.snapmaker.com/t/g-code-solid-preview-alpha/26215/4?u=tyeth

jane-rose, 2022-06-09:
> New Alpha, has preview issue, direction of shell/top line when thin wall (one outer surrounding one middle/shell/top line) see private beta post https://forum.snapmaker.com/t/g-code-solid-preview-alpha/26215/4?u=tyeth

It's a G-Code issue, we will check it.

jane-rose, 2022-06-09:
> Also struggling to get overlap detection working with these models once I add and arrange multiple duplicates, and I also then tried auto-arrange which did a beautiful arrangement, but off the bed and missed one of them (flipped 180 it tucked into the corner and all fit in the snapmaker original square) [plant_pot_risers.zip](https://github.com/Snapmaker/Luban/files/8864232/plant_pot_risers.zip) .

As for this, when there is no place arrange extra model, it will be off the bed.

tyeth, 2022-06-09:
Thanks Jane, I was aware of the off-bed auto arrange issue, but the overlap
detection not working is an issue. Try the auto arrange with a couple of
each of those STLs, it should arrange them easily.