Snapmaker/Luban #1619: Bug: Luban forgets machine IP, does not auto connect via WiFi

HorstBaerbel, 2022-06-10:
# 🐞 bug report

### Affected Version(s)
4.3.0

### Is this a regression? (optional)
Not sure, but I found some related issues

### To Reproduce
Steps to reproduce the behavior:
- I turn the machine on and wait for it to boot up
- I had my Snapmaker previously connected by entering the machines IP. But now the IP-field is empty:
[image]
- When pressing the "+" button the IP is there:
[image]
- Pressing confirm correctly connects the machine
- Now restart Luban

### Exception or Error (optional)
IP field is empty again, Luban does not auto connect, though machine is running:
[image]

### Expected behavior
Luban stores the WiFi IP and auto connects to the machine

## 🌍  Your Environment

**Platform:**
- Operating System: Fedora 36
- Printer: Snapmaker A350T

magicmurven, 2022-06-16:
I also have this issue too :(

polrus, 2022-11-28:
I confirm - the same bug in Linux 4.4.0