Snapmaker/Luban #1622: FEATURE REQUEST: official PPA

erozen, 2022-06-12:
I'd love to be always be running the latest version of Luban, but at present the only option is to manually download and install a .deb file, so automatic updates aren't possible.  I have to remember to periodically check here to see if there is anything new.

Have Snapmaker considered hosting a PPA, so interested people can get updates for Luban as part of a regular apt-get upgrade?  A repo for RPMs would also be appreciated by Rocky/Fedora/Suse/etc users I'm sure.

It can be done using the existing github hosted release files - for example like: https://assafmo.github.io/2019/05/02/ppa-repo-hosted-on-github.html

HorstBaerbel, 2022-06-27:
Please no PPA. Instead I'd vote for a [flatpak](https://docs.flatpak.org/en/latest/getting-started.html) package, so other distributions could benefit from that too! Not everyone uses Debian.

mestadler, 2023-10-06:
I am going to open a FR for flatpak, and to have it published on flathub as a verified app.  Now that the infrastructure is in place  can https://docs.flathub.org/docs/category/for-app-authors.