Snapmaker/Luban #1629: Bug: camera capture incorrect 10W laser

Galindorf, 2022-06-17:
# 🐞 bug report

### Affected Version(s)

 The issue is caused by version 4.3.1

### Is this a regression? (optional)

 Yes, the previous version in which this bug was not present was: the version before this one

### To Reproduce
Steps to reproduce the behavior:
1. Go to camera capture, image is taken but it is blurry/low quality
2. Align cutting lines to image
3. Generate GCode and preview, looks correct
4. Send to workspace, run via Luban
 A clear and concise description of the problem...
The lines cut far off from my alignment via camera capture. Additionally, when the job is finished the work plate stops all the way in the back.[image][image]

The camera capture alignment was working so perfectly in the previous version. I restarted Luban and the laser, loaded all my files again but it did not change the behavior.

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: Windows 10
Snapmaker A350, 10W laser

</code></pre>

Galindorf, 2022-06-17:
[image]

vibutway, 2022-06-17:
From the first picture you provided, the background picture is very obviously skewed; I would like to confirm that you have done the following steps.
1. the camera calibration process is done correctly on the screen side
2. whether the luban control has been calibrated after taking pictures

If the above two steps are not done, it will lead to your printout

Galindorf, 2022-06-17:
OK, I redid my camera calibration (note: this process was working properly just a couple of weeks ago) and am getting the same kinds of issues. Are the images captured by the camera supposed to be so poor quality? Even the calibration image is so bad I can hardly tell where the lines actually are.
[image]

If you zoom in on this image and try to align the lines they are almost impossible to see properly. (In addition, attempting to use a mouse to move the corners tiny amounts is extremely frustrating and difficult!)
 
After a second camera calibration on the touch screen and via Luban, I did a test cut where I used Luban's own shape tool on the calibration page that had been cut.  Luban's vision: 
[image]

Galindorf, 2022-06-17:
[image]

This is how it cut.

vibutway, 2022-06-17:
根据您的反馈，我们对您的步骤进行了一遍覆盖：
1. 在屏幕走了一遍摄像校准
2. luban wifi 连接进行摄像捕捉
3. 对照片进行校准，并应用
4. 然后在背景图添加一个矩形进行切割测试

现象如下：添加矩形的位置
[image]

实际切割出来的成品位置
[image]

看起来一切正常

希望您确认：
1. 在luban校准后已点击应用按钮
2. 拍完照后，机器上的纸张未被移动过位置

vibutway, 2022-06-17:
Based on your feedback, we have overridden your steps to.
1. walked through the camera calibration at the screen
2. luban wifi connection for camera capture
3. calibrated the photo and applied
4. then added a rectangle to the background image for cutting test

The phenomenon is as follows: add the location of the rectangle
[image]

The actual cut out of the finished position
[image]

Everything looks normal

You are expected to confirm that
1. you have clicked the apply button after luban calibration
2. the paper has not been moved from its position on the machine after taking the picture

Translated with www.DeepL.com/Translator (free version)

Galindorf, 2022-06-17:
1. I did click the 'apply' button for sure
2.  I did not move the paper
3. I did this test before when I first installed the 10w laser to do the calibration and it was perfect

I wish to note that this feature WAS working for me in a previous version of Luban. I was using camera capture to cut letters out of wooden craft sticks, which are only 1.5mm wide.

Galindorf, 2022-06-17:
I decided to take video just to make sure I was doing everything correctly and got a different problem: camera capture would not respond. Here is the Video:

https://youtu.be/LLdd0h4wqsw

Galindorf, 2022-06-17:
After restarting the laser and Luban I was able to finish the fresh calibration and cut another test square. The paper was never touched.

Video:
https://youtu.be/HZh7-5eV4EU

Galindorf, 2022-06-17:
Sorry for all the comments! But I got more information. I decided to try another experiment. I noticed that the square I cut seemed to be smaller than the one I had on the screen. 
I made a new document in CorelDrawx8,  8.5x11" (215.9x279.4mm). Rows of circles. I printed the document actual size on my printer and placed on the laser. I used the same file in SVG to cut out the circles. 

Note: The captured image does not appear to be centered. 
[image]

Test1: 
The file imports looking larger than the captured camera image. 
[image]
I was unable to get a perfect alignment between the file and the paper. I sized the file until it lined up as much as possible with the camera image. My paper was very straight on the laser bed, I used a ruler to check it. 
[image]
The cut was completely off: 
[image]

Test 2: 
Placed another printed paper file on the laser bed. This time I did not resize the file in Luban to line it up, I visually centered it as best as I could with the Camera Capture image:
[image]

The cut result was MUCH better:
[image]

Conclusion: the camera capture image is showing the wrong size for the image in Luban.

vibutway, 2022-06-20:
> After restarting the laser and Luban I was able to finish the fresh calibration and cut another test square. The paper was never touched.
> 
> Video: https://youtu.be/HZh7-5eV4EU

Thank you for your feedback, we will try to reproduce it as soon as possible according to the operation you provide

vibutway, 2022-06-20:
We have been able to reproduce your feedback problem, the reason is that the height of the executive head has become high, not the default photo height, from the video you sent over, your executive head height should be installed incorrectly
The video you sent shows the installation position as follows.
[image]

The correct installation position should be
[image]
[image]

Please check it, hope to help you

Galindorf, 2022-06-20:
[image][image]

Mine looks right to me. Also, The camera capture WAS working properly before. Better than it ever did before with the 1.6w laser. I used the camera capture to cut some very small items with accuracy.

Is it possible something is wrong with my camera?

vibutway, 2022-06-21:
I hope you according to the following steps to go through, if the following steps to go through the failure to help you, please contact the aftermarket to change a laser head

1. Please upgrade the screen firmware to the latest version
2. Connect the luban with a serial cable
3. In the console, enter M502 first, then M500
4. reboot the A350 machine
5. Go through the guide process again
6. use luban to do the camera capture

Galindorf, 2022-06-23:
1. My firmware shows as Snapmaker2_V1.14.2_20220113 and says this is the latest version when prompted to look for an update. Is this correct?
2. I am waiting for a long cable to arrive by mail because my computer and laser cannot reach each other.

vibutway, 2022-06-24:
Yes, the latest firmware version is Snapmaker2_V1.14.2_20220113 , so you can start from the second step .

Galindorf, 2022-06-24:
Today I followed steps 2-6. 
My align image was extremely skewed: 
[image]

I did the alignment via Luban but the process is extremely difficult and frustrating to do accurately. 
[image]
I took a new image capture of the camera test and I noticed right away that the whole laser bed does not appear in the Image Capture I took: 
[image]
However! After trying the cut square it was accurate! I was so happy! This is so much better than what happened before. 
[image]

This has not fixed my issue though. I tested another image. 
[image]
You can see the top does not align.  The center does:
[image]
(also, the image quality on this camera is really terrible. Really terrible.)

I made a new file that would be easier to test and took this camera capture: 
[image]
As you can see, this time the camera captured the entire laser bed. However, it is still smaller than the exact file I made to cut it. Restarting the machine did not alter this outcome. I aligned the cut file with my own center mark  on the image capture and got this result: 
[image]

Either my camera is giving a warped view of the laser or the calibration is not working properly? 

I had a different but possibly related weird thing happen, the circles on the Cicada test file did not cut well. Could this all be related to a faulty linear module? 

[image]

vibutway, 2022-06-27:
You can obviously see that your first photo distance is correct, after cutting the distance of the photo again became far away, so it will lead to printing deviations; the following is a luban test package, I hope it can help you.
https://github.com/parachvte/Luban/releases/download/v4.3.2-alpha.2/Snapmaker-Luban-4.3.2-win-x64.exe

Galindorf, 2022-06-27:
I tried the test package with a couple of files and it seems to be working much better now! I cut a couple of squares that came out properly, then I also redid the camera calibration just to make sure. So far it looks like it's doing well, I will do another test tomorrow. Thank  you so much for all of your help! I will let you know how the test goes tomorrow.

ave8tor, 2022-06-28:
I too have been running into a similar issue. I noticed that upon selecting 'Add Background' for the initial scan, the laser goes to the normal X/Y centered position above the work surface, however the Z axis is maxed out, causing the work bed image to be smaller than the work bed itself. A subsequent scan causes the Z axis to lower slightly which ‘mostly’ brings the work bed image into the full frame, however the left side of the scan is not quite to the edge of the work bed. Attached are the images for both the initial and subsequent background image captures.
[image]
[image]
[image]
[image]
Although the image looks better in the sense that the work bed covers most of the drawing area, there is still a sliver on the left that is not to the edge. Furthermore, when I took scans of the wood in the appropriate corners, some of the image was cut off, which also indicates that the scanned area is not calibrated to the dimensions of the work bed. I'll try and provide additional images with a future edit.

I then proceeded to test engrave some rectangles in the four corners and the center of the work bed by tracing the lines to match the picture of the scan. The center rectangle was spot on. At each of the corners however, the etched rectangles were set outside (both on the X and Y axis) from each of the rectangles drawn on the test board.

I have re-calibrated the actual camera with the cut paper several times, as well as ensured the image calibration was accurate prior to each image being added as a background. 

The rectangles were placed as closely to their respective corners as I could get, but it did require me to rotate the test board to do so. The arrow in each box represents the font of the work bed. The individual rectangles are labeled UR/UL/LR/LL/CEN for Upper Right/Upper Left/Lower Right/Lower Left/Center.

[image]

The tool head is attached to the correct anchor points, and the calibration on all settings has been performed with no issues. I have the latest firmware and Luban software. I don’t know if there’s something else I’m not doing (or doing wrong), but I find it frustrating that the camera calibration can’t adjust for the work bed itself. Also, why does it take two separate steps to get a valid image and not center the camera at the correct height on the first iteration?

Galindorf, 2022-07-03:
Suggest: Try printing this file, place on laser bed. Use Camera Capture to take its picture, then load the SVG into Luban and see how it cuts. Then you may see the same issue we are having. (the tiny circle is the exact center of the file)
[image]

I am still having issues. The camera quality on this 10w laser head is so bad. I'm starting to regret it. the 1.6w laser took much better photos. Would it be possible to have the camera capture done the same way as the 1.6w laser, where it takes saveral images and then 'stitches' them together? I know it took longer but the camera quality is so bad I can't even tell what I'm trying to cut. :(

sumibin, 2022-07-04:
There should not be such a large positional deviation in practice. We judge that there is a high probability that it will be a problem with the camera. We will do a verification test on multiple 10W laser heads in these two days.

sumibin, 2022-07-06:
[image]
In testing, the largest deviation we encountered was around 1mm.It looks like everyone's end result is worse than this one.
We will schedule a more advanced and precise calibration in Luban, which may lead to a more complex process to achieve a higher accuracy calibration.
@Galindorf There are two possibilities for the "no round" problem. One is that the workpiece may not be clamped, and an engraving test after clamping the workpiece can verify this problem. The other is that the linear module may be faulty, this needs to contact after-sales support.This also brings problems with Camera Capture

Stringfamily, 2022-07-28:
Hi, I'm running v4.3.2 on MacOS 11.6.8 and have the same camera issues. I know everyone is still having the problem (I assume) but just checking if there is any updates. Thanks!

imh1kiko, 2022-12-05:
I've decided to chime in as well.
My camera captures have been hella blurry from day 1, and my focus is far from what others get, weirdly enough (31.7mm). Same issues as others have stated -- Lines do not match up, and with perfect aligning, there's weird skewing issue as demonstrated above.

tensile23, 2023-04-04:
This issue has expired, please reopen an issue if necessary

closed 2023-04-04 by tensile23

neoscorpio, 2024-04-23:
I have the same issue was this ever solved?