Snapmaker/Luban #1634: Bug: xxxx in 3d printing

tyeth, 2022-07-02:
### Affected Version(s)
4.3.2
 The issue is caused by version x.y.z

### To Reproduce
1. Select Snapmaker Original machine, choose 3d printing, Insert model, setup a TPU profile or select one if you have one.
2. set walls and top+bottom to 1.2mm, layer 0.2mm, initial 0.28mm. Infill 20% with infill as CONCENCTRIC.
3. Look at layers 6-7 or the infil layers, there was no infil. Changing to 25% had some slanted concentric circles (two lines less than 50% of the way around concentric infill area and more like 2% coverage compared to 0% when selecting 20% concentric infill).
4. Change infil to cross and look at same infil layers, they appear reasonable.

## 🌍  Your Environment

**Platform:**
- Operating System: [Windows 11 x64]
- Printer: [Snapmaker Original]
[Blindstopfen_washer.zip](https://github.com/Snapmaker/Luban/files/9033327/Blindstopfen_washer.zip)

jane-rose, 2022-07-11:
@vibutway

sumibin, 2023-03-21:
[image]
The problem still seems to exist in the new engine, keep following up.

closed 2023-03-21 by sumibin