Snapmaker/Luban #1636: Bug: First Layer Print Speed 3mm/s in 3d printing

SirLanceabunch, 2022-07-04:
# 🐞 bug report

### Affected Version(s)

 The issue is caused by version 4.3.0

### Is this a regression? (optional)

 Yes, the previous version in which this bug was not present was: 4.2.8

### To Reproduce
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error
 When printing, the First Layer Print Speed is set to 24mm/s automatically in the selection option but when printing starts the printer will only move at 3mm/s.

Exception or Error (optional)
<pre><code>

 No error given
</code></pre>

Expected behavior
 First Layer Print Speed should not change from what the selection is set at, 24mm/s.

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: Windows 10 x64
- Printer: Snapmaker A350

</code></pre>

bkesty, 2022-07-24:
I am using version 4.3.0 and have the same bug

bkesty, 2022-07-24:
After checking I see that I am on 4.3.2

hollt693, 2022-08-22:
I created a GitHub account just to come here and add my name to the list of people experiencing this problem. I've made some additional observations that may prove helpful in some way. First, the issue doesn't appear to be in the gcode, as I've looked over it myself and also checked it with an online simulator, and in both cases the code seemed to indicate the normal printing speed that I would expect, yet it is realized at 3mm/s on the first layer. Second, I experience the same issue whether I'm running the code on my A350 directly or running it via Luban. Finally, travel moves on the first layer do not appear to be affected, or at least they run at 24mm/s instead of the 3mm/s that printing moves default to.

hollt693, 2022-08-23:
Say, what material profiles are you guys using? The print speed bug happens when I print with TPU, but it seems to be normal when I'm working with PLA.

closed 2023-03-21 by sumibin