Snapmaker/Luban #1637: Bug: Luban won't load or show materials in 3d printing

SirLanceabunch, 2022-07-04:
# 🐞 bug report

### Affected Version(s)

 The issue is caused by multiple versions, 4.3.2, 4.3.1, 4.3.0, 4.2.3, 4.1.4 downloaded them all, same issue on all.

### To Reproduce
Steps to reproduce the behavior:
1. Open Luban 3D Printing (It won't open)
2. Open Luban laser or CNC
3. File>New>3D Printing
4. Materials drop down is empty, Print settings drop down also empty. (See attached screenshot)
5. Luban will not close, must use task manager to force close.

Expected behavior
 Luban should open and provide materials and print styles in the drop downs while allowing changes using the "Settings Gear"

- Operating System: Windows 10 x64
- Printer: Snapmaker A350

SirLanceabunch, 2022-07-04:
[image]

vibutway, 2022-07-05:
Please follow the steps below, it may help you
  1. Open luban
  2. Shortcut key: Shift + Alt + R

SirLanceabunch, 2022-07-05:
This worked! I had to recreate my print settings (possibly laser and CNC settings as well) but that's ok

closed 2022-07-05 by SirLanceabunch