Snapmaker/Luban #1638: Missing submodule SACP-SDK

Andygol, 2022-07-04:
# 🐞 bug report

### Affected Version(s)

 The issue is caused by version x.y.z
4.3.0

### To Reproduce
Steps to reproduce the behavior:
1. clone repo
2. npm install
3. npm run dev
4. See error - `Error: Cannot find module '../../../lib/SACP-SDK/SACP/communication/Dispatcher'`
 A clear and concise description of the problem...
```
10% building 4/4 modules 0 activeℹ ｢wds｣: Project is running at http://localhost:8080/
ℹ ｢wds｣: webpack output is served from 
ℹ ｢wds｣: Content not from webpack is served from /Users/username/Documents/GitHub/Luban/output/app
11% building 13/18 modules 5 active .../username/Documents/GitHub/Luban/node_modules/webpack/hot/log.js(node:65271) UnhandledPromiseRejectionWarning: Error: Cannot find module '../../../lib/SACP-SDK/SACP/communication/Dispatcher'
Require stack:
- /Users/username/Documents/GitHub/Luban/output/server/services/socket/sacp/Business.js
- /Users/username/Documents/GitHub/Luban/output/server/services/socket/sacp/SACP-TCP.js
- /Users/username/Documents/GitHub/Luban/output/server/services/socket/ConnectionManager.js
- /Users/username/Documents/GitHub/Luban/output/server/services/index.js
- /Users/username/Documents/GitHub/Luban/output/server/app.js
- /Users/username/Documents/GitHub/Luban/output/server/index.js
- /Users/username/Documents/GitHub/Luban/output/main.js
- /Users/username/Documents/GitHub/Luban/node_modules/electron/dist/Electron.app/Contents/Resources/default_app.asar/main.js
- 
    at Module._resolveFilename (internal/modules/cjs/loader.js:972:15)
    at Function.n._resolveFilename (electron/js2c/browser_init.js:249:921)
    at Module._load (internal/modules/cjs/loader.js:848:27)
    at Function.f._load (electron/js2c/asar_bundle.js:5:12633)
    at Module.require (internal/modules/cjs/loader.js:1032:19)
    at require (internal/modules/cjs/helpers.js:72:18)
    at Object.<anonymous> (/Users/username/Documents/GitHub/Luban/output/server/services/socket/sacp/Business.js:14:38)
    at Module._compile (internal/modules/cjs/loader.js:1152:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1173:10)
    at Module.load (internal/modules/cjs/loader.js:992:32)
    at Module._load (internal/modules/cjs/loader.js:885:14)
    at Function.f._load (electron/js2c/asar_bundle.js:5:12633)
    at Module.require (internal/modules/cjs/loader.js:1032:19)
    at require (internal/modules/cjs/helpers.js:72:18)
    at Object.<anonymous> (/Users/username/Documents/GitHub/Luban/output/server/services/socket/sacp/SACP-TCP.js:34:33)
    at Module._compile (internal/modules/cjs/loader.js:1152:30)
(node:65271) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 2)
(node:65271) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

The module SACP-SDK is listed in .gitmodules
https://github.com/Snapmaker/Luban/blob/000ea0513b7db707c9e1ffc0f3b49100163e9530/.gitmodules#L4-L6

but there is no repo https://github.com/Snapmaker/SACP-SDK

Expected behavior is being able to run dev version of Luban.

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: [macOS Monterey 12.4, Apple M1 Pro]

</code></pre>

Andygol, 2022-07-10:
It looks like the repository https://github.com/Snapmaker/SACP-SDK has been made private 😞 because GitHub is throwing a 404 error. Could you please make it public again?

jane-rose, 2022-07-10:
> It looks like the repository https://github.com/Snapmaker/SACP-SDK has been made private 😞 because GitHub is throwing a 404 error. Could you please make it public again?

Yep, but we will make it a npm package to solve this issue in the last few days.

Andygol, 2022-07-15:
The issue has been resolved in the PR - https://github.com/Snapmaker/Luban/pull/1654

closed 2022-07-15 by Andygol