Snapmaker/Luban #1640: Bug: Slice failed, model is too small

Koen89, 2022-07-05:
# 🐞 bug report

Affected Version 4.3.2

after updating i can't slice anything anymore.
same print settings in 4.3.1 was working fine.
size of model: 68.4x54.4x52.0

[image]

vibutway, 2022-07-05:
We have recorded your question.
Please follow the steps below, it may help you
(Note that the following steps will backup and clear the original configuration, if necessary, the original configuration can be entered in the settings portal)
1. Open luban
2. Shortcut key: Shift + Alt + R

Koen89, 2022-07-05:
This solved the problem.
thanks!