Snapmaker/Luban #1642: Feature Request: Remember last settings in laser

Galindorf, 2022-07-10:
# 🐞 feature request

### Is your feature request related to a problem? Please describe.

I am making my own very precise presets for some of the materials I use.  When I am doing a lot of work on the same material, I have to scroll through the preset list every time I want to Process my lines. It always defaults to Basswood: Cutting. It wastes a lot of time and is irritating to the work flow. 

### Describe the solution you'd like?

Please have Luban remember the setting you used last and default to it in the "Process" section.  Also, allow to change the order that the Presets show in the list, let us move less important ones to the bottom and more frequently used ones to the top.

jane-rose, 2022-07-11:
@sumibin

jane-rose, 2022-07-15:
It's reasonable, we will add this on subsequent versions of v4.4.0(early august).