Snapmaker/Luban #1646: Feature Request: Filament diameter setting in 3D printing

acanidio, 2022-07-10:
# Feature Request

### Missing filament diameter setting in the material configuration

Even if in the configuration files there is a setting called "material_diameter" this value is not available to be modified in the UI.

### Describe the solution you'd like?

Introduce the field in the material settings

jane-rose, 2022-07-11:
In the subsequent v4.4 version, more than 400 parameters have been opened include the "material_diameter". It will be released on early August.

closed 2022-10-25 by acanidio

reevan8, 2023-07-16:
Hi, I see this issue closed as completed but I am using Luban 4.8.1 and I can't find the material diameter settings anywhere. was it added then removed in a later version? or am I just blind?

captain118, 2023-07-30:
I also do not see it.