Snapmaker/Luban #1665: Progam window using Linux Mint

ipeco85, 2022-07-22:
Hi, as the windows version of Luban (4.3.2) is very slow on my lovely old Sony Vaio notebook I use on my workbench, I'd try using Luban under Linux Mint. 
After some installation troubles caused by my poor Linux knowledge and some hardware restrictions I finally got Luban working. Opening the progam is about 5 times faster as under WIN10. So all could be great, BUT: the Luban desktop doesn't fit into the screen. 
Screen resolution is 1366x768@60Hz. No problem for the Luban win version, but using Linux the Main window is too large for the screen and I cannot scroll to the bottom, especially in the work area. Also SUB Windows for additional adjustments disappear partly at the bottom and cannot be made completely visible - no scolling possible.
Some Linux screen tools help a little, but cannot cure the issue of the missing program window area at the bottom of the screen.
Any ideas for additional adjustments or workarounds?
Thanks
Peter

jane-rose, 2022-07-25:
It may be you had changed the system resolution? One tip, using the main tool bar. Click `Window` and `View in Browser`.

ipeco85, 2022-07-28:
Hi jane-rose, thanks for your response!
I tried several resolutions, but failed. The main window doesn't fit into the screen and in sub-windows no scrolling is possible. The "view in browser" function works for the window size, but has a very poor system response - if any :-(
.. and luban works so fast under linux - what a pity

closed 2023-03-21 by sumibin