Snapmaker/Luban #1679: Bug: Luban in 3d printing when connecting to printer

jenkins007, 2022-08-15:
# 🐞 bug report

### Affected Version(s)

 The issue is caused by version x.y.z

### Is this a regression? (optional)

 Yes, the previous version in which this bug was not present was: ....

### To Reproduce
Steps to reproduce the behavior:
1. Go to Luban Workspace
2. Click on Connect
3. See screen panel that stated Go Home. You must click on OK, then wait for the printer to go home.
4. Most of the time I send the gcode to the printer. Luban should not be on hold to go home. 
 A clear and concise description of the problem...

Exception or Error (optional)
<pre><code>

</code></pre>

Expected behavior
<!-– The program (Luban) should not wait for the printer to go HOME when connecting. If anything, this should happen in the background if nothing else is being printed (although connecting to the printer when printing is disrupting to the active print job. However, 90% of the time, I send the gcode file to the printer. The printer should manage in this situation. If starting on Luban, go home during the print job activity as the starting point. BUT do not pause Luban just to connect. -->

## 🌍  Your E
[image]
nvironment

**Platform:**
<pre><code>

- Operating System: [macOS Monterey]
- Printer: [Snapmaker 350T]

</code></pre>

jane-rose, 2022-09-16:
Due to some limitations of the machine, doing some other operations (such as printing, etc.) when go HOME may cause abnormal machine status. So we define go HOME as blocking behavior.

zeroClearAmerican, 2022-10-06:
I was able to recreate this organically with version 4.4.0. 
I downgraded to v4.3.2 and no longer experience the blocked UI. 

@jane-rose for me, the printer isn't doing anything when Luban is blocked. 
When first connecting to the machine Luban tells the machine to home. After the machine has completed homing and given the "ok" message, the modal does not go away, blocking the rest of the UI and forcing me to restart Luban. 

75% of the time, the exact same thing happens after restarting Luban and/or the machine.

jane-rose, 2022-10-10:
Since i coultn't reproduce it. Please tell me more about it, firmware version and are there any special steps or phenomena? When blocking, is the screen connected to the Internet and luban?

zeroClearAmerican, 2022-11-29:
@jane-rose any update on this? I'd like to be able to upgrade to Luban 4.4.0 for other reasons but I can't even use the machine with 4.4.0 because of this issue.

marhaus, 2023-03-25:
I have the exact same problem with Luban 4.4.0 but on the Snapmaker Original (firmware 2.11).
See screenshot; the console obviously says "OK" from the machine but the "Go Home" window never vanishes.
[image]

sesobebo, 2023-03-29:
Same problem here.
Luban 4.7.0
Snapmaker Original (firmware 2.11)
Windows 10

The printer is connected to PC with a USB cable.
[image]

ChilliSawse, 2023-04-02:
Same issue for me, attempting to connect over USB with a Snapmaker Original. Not only can Luban not recognise the printer I am using but when I input the details I get the same result. 
Asked to home, click okay, but the popup with the spinning circle never goes away. 

[image]