Snapmaker/Luban #1738: Bug: Bed is OFF - Should be ON in 3d printing v4.4.0

tyeth, 2022-09-30:
Not sure how I got this one, but luban 4.4.0 for snapmaker original (it was set to A350 before I reconfigured without closing luban). Selected PETG. Messed with settings, generated gcode, started print via octoprint. 

Problem BED IS OFF.
This is screenshot of gcode header showing desired bed temp and then missing the command from start gcode, and I confirmed in luban that the PETG red has bed set to 70
[image]

tyeth, 2022-09-30:
Reloading Luban and looking back at PETG red for the snapmaker original it now said 40 instead of 70, and I clicked reset and it returned to 70. That doesn't match the previously generated gcode at all.

Generating again with 70 for bed temp it generated with no bed temp instruction. I wonder if it might be an artifact of messing with dual extruder on the A350, as when I goto Material selection page it calls it Left Extruder
[image]

tyeth, 2022-09-30:
[Original-Messing-material_and_print_profiles.zip](https://github.com/Snapmaker/Luban/files/9685195/Original-Messing-material_and_print_profiles.zip)
Material profile, Print profile, model, and generated gcodes

tyeth, 2022-09-30:
I just deleted(moved) my entire roaming profile, so luban setup as new, but same issue when generating code for A350.
[A350_QTPYTop Cover 8mmSlotted_1664552994929.zip](https://github.com/Snapmaker/Luban/files/9685989/A350_QTPYTop.Cover.8mmSlotted_1664552994929.zip)

tyeth, 2022-10-03:
Installing 4.4.0 in Windows Sandbox on another laptop has generated code for the A350 with bed temperature set:
[image]

jane-rose, 2022-10-08:
Try alt+shift+r(Windows), we will check the upgrad issue for original profile.

closed 2023-03-21 by sumibin