Snapmaker/Luban #1740: Bug: Out of memory

loranger, 2022-10-06:
# 🐞 bug report

### Affected Version(s)

Luban 4.4.0

### Is this a regression? (optional)

It seems to be. I never met this issue with previous versions, including 4.3

### To Reproduce
Steps to reproduce the behavior:

Oftenly, with quite simple 3D models, I cannot render any preview and get the `Out of memory. Failed to load preview` which is definitely new to me.

[image]

I use a Macbook air M1 with only 8GB RAM

[image]

[image]

I've never met this issue with the Luban's previous versions. 

### Expected behavior

It should renders the preview as it always did.
For example, after getting the previous error message, I downloaded Luban 4.3.2 and tried with the same STL file, and the same RAM usage (but few seconds later).

It rendered flawlessly

[image]

## 🌍  Your Environment

**Platform:**
<pre><code>
- Operating System: macOS Monterey 12.6
- Printer: Snapmaker 2.0 A350

</code></pre>

loranger, 2022-10-06:
I also own a Macbook Pro M1 Max with 32GB RAM and have the exact same issue whereas there is plenty free ram remaining.

I noticed this happens on this machine (Macbook pro) when I have [luban helper zombie processes](https://github.com/Snapmaker/Luban/issues/1736). When I kill them, I can render preview, at least on my Macbook Pro.

On my Macbook Air, with or without zombie processes, I cannot render models

KP90-SM, 2022-10-09:
I have exactly the same problem Also with MacBook Air with 8Gb memory - but never had a problem previously even with complex files

Only once recently have I been able to render previews - I selected a new material and bizarrely the render previewed when generating gcode after 3 attempts previously when exactly the same model would not render.

I don't think it is a machine problem - but a coding issue with Luban

ckspecials, 2022-10-13:
I to have the exact same problem. I'm using a MacBook Air (M1, 2020, 8GB RAM) running Monterey version 12.6 (21G115)

richiedvd, 2022-10-14:
I have this issue as well, I have three different MAC models... it seems to happen regardless of the MAC configuration or the complexity of the item I'm trying to print.  This is a regression, I am able to successfully preview an item that failed in 4.4.0 in the previous version 4.2.3.  This needs some immediate attention, regressions are unacceptable, is there no QA on this item?

jenkins007, 2022-10-15:
I also have this issue. Using Luban 4.4.0 on an M1 iMac with 16GB RAM. It happens only when there are a large number of items to be printer or Larger models. This happens every time.
The print job still appears to work and I've continued to print after the error message.

44by73, 2022-10-16:
Same issue - Original M1 MacBook Pro - 8GB RAM - MacOS 12.6 - Luban 4.4.0   Also have the zombie helper processes ( https://github.com/Snapmaker/Luban/issues/1736 ) after exiting Luban.   Happens on small and large jobs and on jobs I've run in the past on lower Luban versions.   Happens every time.

DefinityWeb, 2022-10-25:
Same issue with MacBook Pro (16-inch, 2021) - Apple M1 Pro - 16 GB.
This tends to happen with the Smooth Surface preset.

bujordan, 2022-11-06:
Also have this issue, Macbook Air M2, 16 gb ram. Did not ever have this error with previous versions of Luban. Here's a model it has happened several times with: https://www.thingiverse.com/thing:550604 and that renders previews fine in prior versions of luban. This is basically making the current version of luban unusable.

tribals974, 2022-11-19:
i have the same problem on my Mac what can i do to resolve it

ElainaJourney, 2022-11-21:
I am having an identical issue on an Intel based Mac.
[image]
MacBook Pro 2017 2.8 GHz Quad-Core Intel Core i7
16 GB DDR3 RAM

mrJakobsson, 2022-11-26:
I have same issue.
MacBook Air M1 8GB Ventura 13.0.1 Luban 4.4.0

jraVette, 2022-12-03:
Exact same issue for me too.  It seems to be ok for lighter weight slices; i.e., "Fast Print".  but any refinement gives the out of memory error.  Tried on a couple macs and same issue.

[image]

Luban 4.4.0

briancanaday, 2022-12-27:
Same issue

Apple M1 MacBook Pro 8GB Ram macOS: Ventura 13.0.1

matthiasg, 2022-12-28:
Same on M1 Max with 32 GB. though printing still works via luban

golanbenoni, 2023-01-08:
Is there any update to this issue? Experiencing the same on Apple M1 Pro 1running 3.0.1 (22A400).  Interestingly, I don't have this issue while running the same version under Parallels in Windows 11.