Snapmaker/Luban #1758: Bug: Wrong cutting order for laser

daniel-starke, 2022-11-12:
# Wrong cutting order for laser

### Affected Version

Luban 4.4.0

### To Reproduce
Steps to reproduce the behavior:
1. Load an object with an inner path and outer path.
2. Laser cut it.

Expected behavior

The inner path is cut before the outer path so that the whole object does not falls down before all inner parts are cut.

## 🌍  Your Environment

**Platform:**
<pre><code>
- Operating System: Windows 7 x64
- Printer: Snapmaker A350
</code></pre>

parachvte, 2022-11-15:
Inside of an object, paths are distinguished to be inner or outer path.

So if you want specific print order, you can design and separate the two objects, and specify their print order separately.

daniel-starke, 2022-11-15:
I am aware of that workaround.
Nevertheless, if a human can distinguish both, a program is also able to do so.
The easiest way is the [scan line algorithm](https://en.wikipedia.org/wiki/Scanline_rendering).

parachvte, 2022-11-17:
@daniel-starke you are right.

Here is scenarios to be considered,

1. Nested path. We can have more than 2 path embracing each other, you will want to cut from the inner most to outer most path. This can be done by classic geometric algorithms.
2. Overlapping path. If two paths overlap each other, we can not determine which one should be the first to cut; If two path overlap each other, and shaping a closed area, if we want the closed area to be the last to cut, it's a bit difficult to deal with.
3. Open path. It's not a different scenario than 2, both open paths and closed paths can overlap.

To detect overlapping areas, we will need to calculate cross points over scan lines, and generate even more points.
I think we can do simple no-overlapping path detection (like we did for CNC), and ignore overlapping ones.

mutatrum, 2022-12-02:
I ran into the same problem where I cut many small parts from a single sheet. The pathplanner cut off large segments which needed further cutting up, which resulted in misalignments. Furthermore, as the cutting started at the edge, the clamps holding down the plywood didn't serve any purpose anymore.

To solve the scenario's above, one way of solving it would be:
 1. Add intersection points for crossing paths. (optional, crossing paths would be an edge case I assume?);
 2. Start cutting in the center of the work piece;
 3. Only cut off a part that doesn't need any further cutting.

Attached a file I used.

[image]

mutatrum, 2022-12-03:
To have more control over the pathplanner would be highly beneficial. I've found some settings in Lightburn that would work in my case, by using the Order by Group as first priority:
[image]
Hope this helps.

Galindorf, 2022-12-19:
I am adding my voice to this request. I cut lacy works in paper and when the outline is cut first of course the fan blows the pieces out of alignment.  The animated preview that @mutatrum posted is ridiculous... the amount of time wasted on that pathing! Time is money when it comes to laser work, especially if you are cutting things for other people.

end-me-please, 2023-02-04:
similar problem here. it seems to be impossible to cut this part properly.

[image]

viper0078, 2023-04-20:
When importing a DXF file, I would like the cut order to follow the sort order of the DXF file. Or please add an option to specify this behavior. This alone would satisfy my needs.