Snapmaker/Luban #1767: Bug: Job Status : Time Remaining does not update in 3d printing

polrus, 2022-11-28:
When printing from Luban the remaining time does not get updates and shows the total time.
Elapsed time gets updated ok
[image]

### Affected Version(s)
4.4

### To Reproduce
Steps to reproduce the behavior:
1. Start printing in Luban
2. Check remaing time at 0% and at 50% - it is still the same

Expected behavior
Every couple of seconds remaining time should decrees.

**Platform:**
<pre><code>

- Operating System: Opensuse Tumbleweed
- Printer: A350

</code></pre>

timpursell, 2023-01-01:
I'm experiencing this exact same issue on the same machine model and software version.

timpursell, 2023-04-15:
Great news - this seems to be fixed in the latest version (4.7.2)

[image]